qmk/qmk_firmware #22457: [Bug]

BionicRiddle, 2023-11-12:
### Describe the Bug

This is somewhat of a contionuation on the issue #22269. A couple of days ago after merging the master branch into my personal repository. Following this update, the RGB on the slave half failed to turn off when my computer shut down.

Upon investigation, I identified that the change in issue #21287 was the cause of this problem. To resolve the issue temporarily, I reverted the change.

My setup uses an rp2040 (Liatris), and for master/slave detection, I use the following config:
`#define	SPLIT_HAND_PIN_LOW_IS_LEFT
#define SPLIT_HAND_PIN D4`

Referencing @drashna's insights in issue #22269:
> In theory, that shouldn't matter, as suspend is never called on the slave side, and rgb matrix syncs the suspend state, so it should be turning off the rgb.

For some reason the rgb (and OLED) state does not get synced. 

### Keyboard Used

sofle/rev1

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/riddle/qmk_firmware
Ψ Detected Linux (WSL, Debian GNU/Linux 12 (bookworm)).
Ψ Git branch: master
Ψ Repo version: 0.22.14
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2023-11-11 23:17:57 +0100 (c5e7039aff) -- FFJHS
Ψ - Latest upstream/master: 2023-11-12 06:30:31 +0100 (c2326c05c0) -- [Keyboard] Fixes to make snes_macropad play nicer with qmk configurator (#22432)
Ψ - Latest upstream/develop: 2023-11-11 05:55:47 -0500 (2e1a27da60) -- fix: make clicky delay silent (#21866)
Ψ - Common ancestor with upstream/master: 2023-11-11 20:50:33 +1100 (211fbbd16d) -- Add `qmk license-check` developer-level CLI command. (#22075)
Ψ - Common ancestor with upstream/develop: 2023-11-11 20:50:33 +1100 (211fbbd16d) -- Add `qmk license-check` developer-level CLI command. (#22075)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 7.1
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

GilDev, 2023-11-23:
Same behavior on my keyboard derived from a Sofle Choc. Will try the fix mentioned in the linked issue.

l3ztum, 2024-02-08:
tried the [fix](https://github.com/qmk/qmk_firmware/issues/22269#issuecomment-1807105115) worked on my split keyboard so far

EmilSodergren, 2024-06-17:
This solved my issue on my lily58 aurora with liatris boards. Please PR this!

PR #24020 (2024-07-04): Change suspend condition check order on ChibiOS
## Description

Specifically, it seems on some systems that the sync doesn't happen correctly, and changing the order should address the issue by always calling `suspend_wakeup_condition`, which calls `matrix_scan`

I haven't experienced this issue, so I can't really test it.

## Types of Changes

- [x] Core
- [x] Bugfix

## Issues Fixed or Closed by This PR

* Closes #22269 
* Closes #22457

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/ [... cut]