qmk/qmk_firmware #22477: [Bug] STM32F401 needs reboot with a powered usb hub when the host is disconnected

coliss86, 2023-11-14:
### Describe the Bug

Hi,

I built myself a dactyl with a STM32F401. I followed the configuration of [tractyl_manuform/5x6_right/f411](https://github.com/qmk/qmk_firmware/blob/master/keyboards/handwired/tractyl_manuform/5x6_right/f411/mcuconf.h).

This keyboard is connected to a self powered hub USB. Whenever I disconnect my mac connected from this hub and plug it back, the keyboard is not recognized any more. I have to disconnect/reconnect the USB cable from the keyboard in order to bring it back.
I do not have this problem with a macro pad based on a bluepill, nor with a Ducky One 2.

Is it a known problem ? What can I try to do ?

Thanks

### Link to product page (if applicable)

https://gitlab.com/coliss86/dactyl/-/tree/main/qmk

### Operating System

mac os

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [X] Karabiner (macOS)

drashna, 2023-12-03:
Yeah, this is a known issue and in part due to the split usb detection.  There is a mod for adding vbus detection, but requires soldering onto a very small diode.  

https://discord.com/channels/440868230475677696/440911374571274240/876910618995740732
and
https://discord.com/channels/440868230475677696/440911374571274240/876912742974496829

However, it is possible that this is not related to that, and may be an issue with the usb endpoint itself. If so, then this may help:
https://github.com/qmk/qmk_firmware/pull/21656

closed 2023-12-04 by coliss86

coliss86, 2024-02-10:
Time flies and I didn't try the above solution but I came across a similar [issue suggesting](https://github.com/qmk/qmk_firmware/issues/22778#issuecomment-1872614141) trying to set `#define SPLIT_WATCHDOG_ENABLE` in config.h : it works like a charm !