qmk/qmk_firmware #22478: [Bug] Tidbit's TRRS doesnt work

freemo, 2023-11-15:
### Describe the Bug

When I connect my snap to my tidbit via the right-most trrs port the tidbit powers up with static on the oled but otherwise doesnt work. Keypresses arent recorded. Both keyboards work fine when connected directly.

Here is the keymap/firmware I use on the snap.

https://git.qoto.org/freemo/qmk-firmware/-/tree/freemo/keyboards/nullbitsco/snap/rp2040/keymaps/freemo

Here it is for the tidbit:
https://git.qoto.org/freemo/qmk-firmware/-/tree/freemo/keyboards/nullbitsco/tidbit/keymaps/freemo

As you can see the QMK app seems to have created a slightly different directory structure but I dont think that makes a difference.

Only other thing I tried so far was to turn on "SPLIT_KEYBOARD" in the rules for the tidbit. If I do that, however, it wont compile. I saw from other posts that this should "just work" so im not really sure what is going on here... Any ideas?

### Keyboard Used

nullbitsco/tidbit

### Link to product page (if applicable)

https://nullbits.co/tidbit/

### Operating System

NixOS - Linux

### qmk doctor Output

```text
❯ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/freemo/source/qmk_firmware
Ψ Detected Linux (NixOS 23.11 (Tapir)).
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'wb32-dfu' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: freemo
Ψ Repo version: 0.22.14
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest freemo: 2023-11-13 01:17:53 -0500 (42e583a767) -- Added initial keymap for my tidbit
Ψ - Latest upstream/master: 2023-11-09 04:03:07 +0200 (daabe2d8c5) -- Update layer state of Lily58 from 'Default' to 'Base' (#22397)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-11-09 04:03:07 +0200 (daabe2d8c5) -- Update layer state of Lily58 from 'Default' to 'Base' (#22397)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

keyboard-magpie, 2023-11-15:
The remote keyboard that @jaygreco implemented will only work with AVR (not rp2040), to the best of my knowledge. Perhaps best discussed with nullbits directly on their discord https://discord.gg/K9fm7ndzsG

jaygreco, 2023-11-15:
I saw that there was a Reddit post with this issue as well. There are two things here, which I'd like to address separately: 

1. The SNAP + RP2040 fully supports TRRS, and it should just work, but needs to be connected to the same side as USB is. TRRS to a TIDBIT won't work over the split TRRS connection, if that makes sense. 

1. Upstream support Bit-C PRO RP2040 on our boards hasn't been added yet. We have a nullbits fork + branch with all RP2040 support [here](https://github.com/jaygreco/qmk_firmware/tree/rp2040_clean/keyboards/nullbitsco), including the SNAP and TIDBIT, but I haven't had the energy to open and push through a PR. TRRS for these platforms falls under this, if that makes sense. Our precompiled firmware (which uses the branch above) supports these out of the box.

freemo, 2023-11-15:
I will link to the reddit in case it is helpful to anyone (as I did reply to advice there):

https://www.reddit.com/r/nullbits/comments/17veqa5/snap_tidbit_via_trrs_not_working/

freemo, 2023-11-16:
So had a partial breakthrough. The issue is a regression.

When I tried using the official firmware those are all nightly builds, and thus why they broke, there was a regression issue at some point. For the AVR nullbit has actual releases, not nightly builds. So I checked out the same commit used for AVR and used that to build the firmware and TRRS. It worked but broke many other things (timers and the encoder, caps and numlock reports, and layers arent shared). But the TRRS itself works. Obviously ideally I'd like to get this working on newer code where all the other stuff isnt broken still.

Here is my branch for the above code where TRRS is working. Here is the keymap/firmware I use on the snap.

That said I cant figure out why the original code is broken still... Its something very fundemental as it doesnt even fully initialize when powered via trrs.

https://git.qoto.org/freemo/qmk-firmware/-/tree/freemo-trrs/keyboards/nullbitsco/snap/rp2040/keymaps/freemo

Here it is for the tidbit:
https://git.qoto.org/freemo/qmk-firmware/-/tree/freemo-trrs/keyboards/nullbitsco/tidbit/keymaps/freemo

jaygreco, 2023-11-16:
I'm extremely confused. Are you on AVR or RP2040? 

GH issues probably isn't the right place to discuss this either. This isn't an upstream QMK issue.

freemo, 2023-11-16:
> I'm extremely confused. Are you on AVR or RP2040?

RP2040. I used the QMK firmware as it was when the AVR based firmwares were released but compiled for RP2040. There are no official firmwares compiled for RP2040 just the nightly which didnt work.

> 
> GH issues probably isn't the right place to discuss this either. This isn't an upstream QMK issue.

It appears to in fact be upstream QMK which is why I updated here. The reason I say that is because both the nightly and code compiled off the current upstream has this problem. While code compiled off an earlier version of upstream works. Therefore the problem appears to be an issue introduced in upstream QMK and not an issue for the nullbit fork which does work (but outdated with other issues as mentioned).