qmk/qmk_firmware #22494: [Bug] cannot build keyboards/hhkb/yang with `BLUETOOTH_ENABLE`

orumin, 2023-11-18:
### Describe the Bug

keyboards/hhkb/yang is code for HHKB with modified version controller to support Bluetooth Low Energy.
but this code cannot build with `BLUETOOTH_ENABLE = yes` since this code is merged into master.

the code for this keyboard calls no exist API to control Adafruit UART friends, I guess.

### Keyboard Used

hhkb/yang

### Operating System

WSL/Ubuntu 22.04

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/orumin/qmk_firmware/hhkb_yang_orumin
Ψ Detected Linux (WSL, Ubuntu 22.04.3 LTS).
⚠ QMK home does not appear to be a Git repository! (no .git folder)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb3)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

`qmk doctor` reports warning of git, this is expected behavior because I use `git-worktree(1)` to checkout code from multiple branch at same time.
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

fauxpark, 2023-11-18:
The HHKB Bluetooth code was removed due to bitrot.

closed 2023-11-18 by fauxpark

orumin, 2023-11-18:
@fauxpark you say about version of RN42 controller. YANG's mod wan't removed

orumin, 2023-11-18:
https://github.com/qmk/qmk_firmware/tree/master/keyboards/hhkb/yang here is code.

orumin, 2023-11-18:
@fauxpark could you re-open this issue? I tried to create PR to resolve this issue.
to resolve issue, maintain above linked code, or remove those of code, I think

fauxpark, 2023-11-18:
The Bluefruit LE code needs some serious rework first which most likely will not incorporate any of these changes.

orumin, 2023-11-18:
If so that, remove BLE and related functions from feature list in `keyboards/hhkb/yang/readme.md` will be better than resolve this issue with no change.
this description make confuse and loss time other users.