qmk/qmk_firmware #22501: Keyboard error when rotating encoder（旋转编码器时键盘错误）

biuboom-jun, 2023-11-19:
### Issue Description

Other functions of the keyboard are normal, and pressing the knobs is normal. Forward or reverse rotation will cause the program to get stuck. After outputting the correct function once, it immediately got stuck and other buttons or functions became ineffective. Power on the keyboard again and return to normal. code show as below:
键盘其他功能正常，按下旋钮也正常。 正转或反转都会导致程序卡住。输出一次正确的功能后，立即卡住，其他按钮或功能失效。 键盘重新上电，恢复正常。 代码如下：
MCU: STM32F401RC

//rules.mk
ENCODER_ENABLE = yes

//config.h
/* Encoder 旋转编码器*/
#ifdef ENCODER_ENABLE
#define ENCODERS_PAD_A { A6 }
#define ENCODERS_PAD_B { A7 }
#define ENCODER_RESOLUTION 4
#endif

//keymap.c
/*Encoder Variables  旋钮*/
#ifdef ENCODER_ENABLE
bool encoder_update_user(uint8_t index, bool clockwise) {
    if (index == 0) { /* First encoder */
        if (clockwise) {
            tap_code(KC_VOLU);
        } else {
            tap_code(KC_VOLD);
        }
    }
    return false;
}

drashna, 2023-12-10:
Is this a blackpill controllers?

biuboom-jun, 2023-12-12:
> Is this a blackpill controllers?

The problem has been resolved. Placing the encoder configuration in info.json will eliminate this issue

closed 2023-12-12 by biuboom-jun