qmk/qmk_firmware #22510: [Bug] qmk new-keyboard fails to compile ChibiOS for RP2040 and split layout

whitehorsesoft, 2023-11-20:
### Describe the Bug

Errors during compilation when using `qmk new-keyboard`  followed by `qmk compile` for that new keyboard.

`qmk new-keyboard` uses split_3x5_2 layout, and RP2040 microcontroller.

### Keyboard Used

Any

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: [obscured]/qmk
Ψ Detected Linux (Arch Linux).
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp [obscured]/qmk/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp [obscured]/qmk/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp [obscured]/qmk/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp [obscured]/qmk/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp [obscured]/qmk/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'wb32-dfu' boards. Run 'sudo cp [obscured]/qmk/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp [obscured]/qmk/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp [obscured]/qmk/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp [obscured]/qmk/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp [obscured]/qmk/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp [obscured]/qmk/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp [obscured]/qmk/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: master
Ψ Repo version: 0.22.14
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.2.0
Ψ Found avr-gcc version 13.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.0.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb16)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb3)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

keyboard-magpie, 2023-11-20:
You haven't actually included the compilation errors. 
Its worth noting that `qmk new-keyboard` just generates very vague code that you will need to edit; it won't work directly.

Your `qmk doctor` error output can be remedied by running `make git-submodule` in the `/qmk_firmware` directory.

whitehorsesoft, 2023-11-20:
Full error file from `qmk compile` attached. Start of error looks like this:

> `Ψ Compiling keymap with make --jobs=1 test:default`
`In file included from ./lib/chibios/os/hal/include/hal_pal.h:174,`
`                 from ./lib/chibios/os/hal/include/hal.h:306,`
`                 from platforms/chibios/gpio.h:18,`
`                 from platforms/gpio.h:21,`
`                 from quantum/matrix.h:22,`
`                 from quantum/matrix.c:21:`
`platforms/chibios/_pin_defs.h:129:21: error: 'GPIOD' undeclared here (not in a function)`
`  129 | #define D1 PAL_LINE(GPIOD, 1)`
`      |                     ^~~~~`

[error.log](https://github.com/qmk/qmk_firmware/files/13412114/error.log)

keyboard-magpie, 2023-11-20:
rp2040 pins are defined as `GPx` eg `GP1`

whitehorsesoft, 2023-11-20:
> Its worth noting that qmk new-keyboard just generates very vague code that you will need to edit; it won't work directly.

@keyboard-magpie thanks, I didn't pick up this skimming through the [porting keyboards documentation](https://docs.qmk.fm/#/porting_your_keyboard_to_qmk) the first time. Closing this bug!

closed 2023-11-20 by whitehorsesoft