qmk/qmk_firmware #22514: [Bug] A potential bug may exist in usb_lld_init_endpoint in ChibiOS in melgeek firmware

XinDistance, 2023-11-20:
### Describe the Bug

In function usb_lld_init_endpoint, it reads a device register value and as an offset to fetch data. However, this value is not checked thus result in arbitrary memory read. This is a problem caused by ChibiOS. I don't know how to fix it.

### Keyboard Used

melgeek

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

XinDistance, 2023-11-20:
And another one in the interrupt handler VectorB0(). The function pointer is controlled by a device register whose memory address is 0x40000024.

sigprof, 2023-11-20:
Your report is really vague:
- `melgeek` is not a single specific keyboard (it's a vendor folder which has lots of keyboards inside, and those keyboards even use different MCUs);
- `usb_lld_init_endpoint()` has multiple implementations (there are many LLDs for various MCUs); not sure which one do you mean;
- can you point to the specific line in the source which has the problematic access?

Also `0x40000024` is apparently the address of the `TIM2->CNT` register on most STM32 MCUs; not sure why any code would use its value in pointer calculations directly.