qmk/qmk_firmware #22540: JJ40 v2.0 on QMK ?

papourien, 2023-11-25:
### Issue Description

hello

I bougth a JJ40v2 and accidently earase the original firmware with my old 1.5 firmware
now the pcb dont respond any more except flashing it but i don't found a compatible firmware for this v2
the file.hex on via dont works on this v2
some help ? maybe a qmk support soon ?

sigprof, 2023-11-25:
Unfortunately, the JJ40 V2 PCB is not supported (in general, KPrepublic does not care about submitting the firmware sources for their boards to the upstream QMK, and most of the existing support had been added by volunteers, sometimes even by reverse engineering the PCBs).

You may try asking the seller for a firmware file (KPrepublic did not publish the firmware for that PCB at http://kpchn.com/s/PoHJ).  In theory they are supposed to provide also the complete source code (not just a compiled firmware file) if they are actually using the QMK source code.

The only other option would be reverse engineering the PCB (assuming that it actually has a compatible MCU, but that seems likely, because otherwise you probably would not be able to flash the old .hex file).

papourien, 2023-11-25:
thx sigprof...next time i'll ask before flashing
the shop on ali-express can't provide me the firmware, i'll ask to the web site kprepublic support directly...
if not, can i "extract" the firmware from a new pcb to restore mine ?

sigprof, 2023-11-25:
If the PCB uses ATmega32u4 with the `atmel-dfu` bootloader, you can't extract the firmware over USB — the bootloader does not allow reading the existing firmware.  Reading the flash over the ISP connection may be possible, but that requires extra hardware, and the PCB might not even have the ISP breakout.

Can you look at the back side of the PCB that you have — does the MCU chip have some readable markings?  You may also share some photos of the PCB (apparently KPrepublic did not even bother to update their photos; do they at least have something saying V2 there?).

papourien, 2023-11-25:
[image]
[image]
[image]
[image]
[image]

sigprof, 2023-11-25:
So apparently there is no ISP breakout on this PCB, and even worse, at least some of the ISP pins don't seem to be connected to anything, so if this PCB ever requires ISP flashing due to some bootloader corruption, it would require connecting wires directly to chip pins on the QFN package.  At least the 5.1k resistors on CC pins seem to be present, so this PCB should work correctly with C-to-C cables (some other PCBs still have errors in this area).

Making a QMK firmware for this PCB should be possible with some reverse engineering effort, but KPrepublic should really fulfill their obligations to provide the complete source code for the firmware…

papourien, 2023-11-25:
thx for all this informations, the pcb respond to the QMK toolbox with the reset buton, i can flash it again.
I'll wait the response of kprepublic...

papourien, 2023-11-28:
KP republic don't care of my problem :(
any advise for a PCB 40% ortho QMK compatible ?
somebody want's my jj40v2 ? for some rescue or retro ingenering ?

papourien, 2023-12-18:
hi here, using https://kbfirmware.com and with some retro engineering, i got back 10 columns and 3 raws...
I'll check my pins again to be sure...it's small...
how i check the backlight rgb pin ? need other pins ?

papourien, 2023-12-29:
ah, i got the file_via.hex

source 
http://kpchn.com/s/JerSP?spm=a2g0s.imconversation.0.0.4f3a3e5fMeZWUq

mirror
https://drive.google.com/file/d/1Su-q29TAwEOQfJyy765lMkvDX2I8eSax/view?usp=drive_link

I get back the rgb and the via support
but still have shortcut on 2 raws and 2 columns

closed 2024-04-28 by github-actions[bot]