qmk/qmk_firmware #22564: [Feature Request] Support for Keychron q5 Pro Knob Asani

1-lost-cause, 2023-11-29:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

As mentioned in the guide, this is a request for Keychron Q5 Pro Knob Asani

adophoxia, 2023-11-29:
If this is meant to be asking "Can you please add the Keychron Q5 Pro into the main QMK repo", the answer is "very unlikely". You can however work off of Keychron's fork of QMK, specifically their [`bluetooth_playground`](https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground) branch that has the Q and K Pro boards.

zvecr, 2023-11-29:
As part of the template you filled in:

> Please refrain from asking maintainers to add support for specific keyboards -- it is unlikely they will have hardware available, and will not be able to help.
Your best bet is to take the initiative, add support, then submit a PR yourself.

closed 2023-11-29 by zvecr