qmk/qmk_firmware #22590: [Feature Request] Add output file argument to `qmk compile`

leah-splitkb, 2023-12-01:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

Prior to #22221, the command `qmk compile -e CONVERT_TO=liatris --keyboard splitkb/kyria/rev3 --keymap default` would produce the file `splitkb_kyria_rev3_default_liatris.uf2`. After #22221, the same command would produce the file `splitkb_kyria_rev3_default.uf2`.

Although a minor change and easily resolved, this change _did_ break our CI pipeline and as far as I can tell wasn't documented.

It would be neat if there was a way to do `qmk compile (..) --output firmware.uf2`. This would get rid of the whole "guess what filename QMK is going to use" part, and make it a lot easier to set up reliable CI pipelines.

drashna, 2023-12-10:
`-e TARGET=(name)` will do that

tzarc, 2024-04-21:
#22887 should sort this out.

PR #22887 (2024-08-12): Userspace: add support for adding environment variables during build
## Description

A few complaints on Discord and the like for not being able to specify things like `CONVERT_TO` and `FORCE_LAYOUT` when using userspace.

This adds support for the usual `-e`/`--env` parameters to `qmk userspace-add` and `qmk userspace-remove`, and ensures such extra environment variables are propagated across to each build.

Userspace repo version is bumped to `1.1` as a result of the addition to the file format.

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
-  [... cut]