qmk/qmk_firmware #22596: Compiling from `qmk_firmware` will clone binaries to `qmk_userspace`

adophoxia, 2023-12-02:
### Information

- [X] None of the other issue templates apply

### Description

When working on some PRs over in `qmk_firmware`, despite my `qmk.json` being configured to include only my Keychron Q2 and Q0 (`base`) as shown here:
```json
{
    "userspace_version": "1.0",
    "build_targets": [
        ["keychron/q2/ansi_encoder", "adophoxia"],
        ["keychron/q0/base", "adophoxia"]
    ]
}
```
when I compiled for `qmk compile -kb keychron/q5/ansi -km default` for testing earlier, the binary for it will show up in both forks' directories. It was only there until I clean the `qmk_userspace` directory with `qmk clean --all`.

(`qmk_userspace` - `qmk doctor` output)
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/qmk/qmk_firmware
Ψ Detected Linux (WSL, Debian GNU/Linux 11 (bullseye)).
Ψ Testing userspace candidate: /home/qmk/qmk_userspace -- Valid `qmk.json`
Ψ Testing userspace candidate: /home/qmk/qmk_userspace -- Valid `qmk.json`
Ψ QMK userspace: /home/qmk/qmk_userspace
Ψ Userspace enabled: True
Ψ Git branch: personal
Ψ Repo version: 0.23.0
Ψ - Latest personal: 2023-12-01 21:47:31 -0800 (3a15f0d1ad) -- Encoder Enhancements
Ψ - Latest upstream/master: 2023-12-02 01:05:46 +1100 (afa0eab374) -- Fix a bunch of CLI warnings (#22589)
Ψ - Latest upstream/develop: 2023-12-01 14:06:15 +0000 (07e2b645ab) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2023-12-02 01:05:46 +1100 (afa0eab374) -- Fix a bunch of CLI warnings (#22589)
Ψ - Common ancestor with upstream/develop: 2023-12-02 01:05:46 +1100 (afa0eab374) -- Fix a bunch of CLI warnings (#22589)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.3.0
Ψ Found avrdude version 6.4
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
```

tzarc, 2023-12-02:
Had a quick discussion internally; given the long-term vision is to abstract/hide away `qmk_firmware` for 99% of people (i.e. people not making any sort of core/keyboard changes), we don't really think this is an issue.

Are there particular reasons you _don't_ want them duplicated?

adophoxia, 2023-12-03:
For "consistency", being how I'd like it to only include binaries from boards that I've added in `qmk.json` and to exclude everything else. Also kind of a PITA when on `qmk_firmware`, you're trying to test some kb changes for non-configured boards and when you swap back to `qmk_userspace`, you see them show up and go "How the f did they get in here?"

bcat, 2024-01-07:
It seems the reverse is true as well... compiling from `qmk_userspace` results in redundant output files in the `qmk_firmware` directory. I ran into this when [digging into a different issue on the Discord](https://discord.com/channels/440868230475677696/1146201809224994887/1193411428825190500).

It's a super minor thing, but unintuitive nevertheless. I would expect build output to go to the current directory (and nowhere else) unless otherwise specified. A bit of a principle of least surprise violation. :)

bcat, 2024-04-27:
Not stale; this still happens. (Of course, maybe it's considered WAI.)

tzarc, 2024-04-27:
Indeed, NAI.

closed 2024-04-27 by tzarc