qmk/qmk_firmware #22611: [Bug] KRepublic JJ50 Config out of date?

damouse, 2023-12-05:
### Describe the Bug

I just bought a JJ50 and tried to flash QMK on it. Unfortunately, no keys or lights work after flashing the default layout using QMK Configurator. [The config](https://github.com/qmk/qmk_firmware/blob/master/keyboards/kprepublic/jj50/info.json#L42) lists the microprocessor as `atmega32a`, but looking at the back of the PCB the microprocessor reads `atmega32u4`. Is the QMK config out of date? If so, what's the process for updating this? Am I doing something else wrong? 

### Keyboard Used

krepublic/jj50

### Operating System

ubuntu

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

fauxpark, 2023-12-05:
The JJ50 PCB looks like this:
https://kprepublic.com/collections/jj50-50/products/jj50-50-custom-keyboard-pcb-similar-with-preonic
You do not have a JJ50.

damouse, 2023-12-05:
Here's a larger picture: [pic](https://imgur.com/a/vaOxXog).

fauxpark, 2023-12-05:
Seems like they've made a new version then, but did not bother to submit it here. That's pretty typical. You'll have to ask them for help.

damouse, 2023-12-05:
I'll get in touch with krepublic.

sigprof, 2023-12-05:
Looks like the same issue as with JJ40 (#22540) — KPrepublic made a new PCB revision and did not bother to publish any kind of firmware sources.  This time they apparently did not even bother to publish a compiled firmware .hex, so if you flashed some wrong firmware to the new PCB revision, you don't even have a way to restore a working firmware.

Note that the back side of the PCB has the “JJ50_V2” marking:
[image]

papourien, 2023-12-05:
welcome, i'll do the same with the jj40v2...
if my seller give me a solution for the jj40, i would ask for the jj50 too

damouse, 2023-12-05:
That is exactly what I did. I thought the board was bricked, as none of the
flashing utilities recognized the board, and I couldn't see it in USB
debugging tools.

However, I found a small button on the back of the PCB next to the
microprocessor. Pressing this and plugging it in caused the board to show
up in qmk_toolbox. I was able to flash the (wrong) firmware again, so I
assume the board is not bricked, but I'll wait until I get working firmware
to say for sure.

damouse, 2023-12-19:
Fixed in https://github.com/qmk/qmk_firmware/pull/22713.

closed 2023-12-19 by damouse