qmk/qmk_firmware #22631: [Bug] `CONVERT_TO` RP2040 targets broken with `UART_DRIVER_REQUIRED=yes`

jaygreco, 2023-12-08:
### Describe the Bug

I'm fixing up the Bit-C PRO RP2040 converters and have run into an issue with `UART_DRIVER_REQUIRED`, which has been introduced in the [latest breaking change](https://github.com/qmk/qmk_firmware/blob/master/docs/ChangeLog/20231126.md?plain=1#L125) merge. All extra source included in `rules.mk` fail to compile with the following error:

```
Compiling: keyboards/nullbitsco/nibble/matrix.c                                                    In file included from ./lib/chibios/os/hal/include/hal.h:319,
                 from platforms/chibios/platform_deps.h:18,
                 from quantum/quantum.h:18,
                 from keyboards/nullbitsco/nibble/matrix.c:16:
./lib/chibios/os/hal/include/hal_serial.h:99:10: fatal error: hal_serial_lld.h: No such file or directory
   99 | #include "hal_serial_lld.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
 [ERRORS]
```

### Keyboard Used

nullbitsco/nibble

### Operating System

Ubuntu 22.04

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/ubuntu/firmware/qmk_firmware
Ψ Detected Linux (Ubuntu 22.04.2 LTS).
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/ubuntu/firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/ubuntu/firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/ubuntu/firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/ubuntu/firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/ubuntu/firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'wb32-dfu' boards. Run 'sudo cp /home/ubuntu/firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/ubuntu/firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/ubuntu/firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/ubuntu/firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/ubuntu/firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/ubuntu/firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/ubuntu/firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.23.1
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2023-12-08 09:55:24 +0000 (0f341dfdd3) -- Fix TARGET when using converters (#22591)
Ψ - Latest upstream/master: 2023-12-08 09:55:24 +0000 (0f341dfdd3) -- Fix TARGET when using converters (#22591)
Ψ - Latest upstream/develop: 2023-12-08 15:57:34 +0000 (42245651e7) -- Align Dip Switch feature (#22625)
Ψ - Common ancestor with upstream/master: 2023-12-08 09:55:24 +0000 (0f341dfdd3) -- Fix TARGET when using converters (#22591)
Ψ - Common ancestor with upstream/develop: 2023-12-08 09:55:24 +0000 (0f341dfdd3) -- Fix TARGET when using converters (#22591)
Ψ CLI installed in virtualenv.
☒ Can't find avrdude in your path.
Would you like to install dependencies? [Y/n] m
Would you like to install dependencies? [Y/n] n
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 8.2.0
⚠ Unknown version for avrdude
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb16109)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ Major problems detected, please fix these problems before proceeding.
Ψ Check out the FAQ (https://docs.qmk.fm/#/faq_build) or join the QMK Discord (https://discord.gg/Uq7gcHh) for help.
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

The converter updates are minor changes to resolve missing LED pins during compilation.

sigprof, 2023-12-08:
The real problem here is that `nullbitsco/nibble` is one of the rare boards which uses the QMK UART API (`uart.h`), but the `uart.c` implementation for ChibiOS supports only the `SERIAL` driver, while the RP2040 port provides only the `SIO` driver.  Because of that problem using a RP2040-based controller currently won't work for that board, unless some option to disable the custom “remote keyboard” code is added.  The `UART_DRIVER_REQUIRED` change just makes things fail earlier (before that change the compile would fail only on `uart.c`).

The proper way to fix things would be to implement support for the `SIO` backend in `uart.c`, and then adjust the code which handles `UART_DRIVER_REQUIRED` to enable either `HAL_USE_SERIAL` or `HAL_USE_SIO` as appropriate for the MCU (currently that code blindly enables `HAL_USE_SERIAL`, which then fails on MCUs which don't have the `SERIAL` driver).  Alternatively, someone could add a `SERIAL` driver for RP2040 to ChibiOS-contrib, so that the existing `uart.c` code could be used.

sigprof, 2023-12-08:
> I have the serial working with the raw SIO driver in [a dedicated branch](https://github.com/jaygreco/qmk_firmware/blob/rp2040_clean/keyboards/nullbitsco/common/remote_kb.c) that we use for our CI/precompiled builds.

Code like that is probably not suitable for the upstream QMK as is — it should be moved to `platforms/chibios/drivers/uart.c`, so that the UART API would be available for all keyboards.

> That seems enough to add with a preprocessor check and was already have stubbed it out so that the boards compile with the converter in my working branch; happy to explore it further if it's a potential workaround for this.

With the current implementation of `UART_DRIVER_REQUIRED` the option would need to be at the makefile level (probably `post_rules.mk`, because only simple assignments are allowed in `rules.mk`) — the `UART_DRIVER_REQUIRED = yes` setting would need to be omitted to avoid errors due to this code:
```make
ifeq ($(strip $(UART_DRIVER_REQUIRED)), yes)
    OPT_DEFS += -DHAL_USE_SERIAL=TRUE
    QUANTUM_LIB_SRC += uart.c
endif
```

Enabling `HAL_USE_SERIAL` if the MCU does not actually support that driver would result in errors.  Probably `mcu_selection.mk` should set something like `CHIBIOS_SERIAL_DRIVER_SUPPORTED = no` for RP2040, and then the code handling `UART_DRIVER_REQUIRED` would need to set `OPT_DEFS += -DHAL_USE_SIO=TRUE`.

cookyt, 2023-12-21:
Hey, this issue is affecting me as well. I can't compile qmk for my nullbitsco/snap keyboard because I'm using a Bit-c Pro (which is RP2040-based).

What're the next steps here? It seems like the fundamental issue is that `qmk_firmware/platforms/chibios/drivers/uart.{c,h}` need to be re-worked to use HAL_SIO if HAL_SERIAL isn't supported by the underlying MCU. Is that correct? If so, how can I help?

vinorodrigues, 2023-12-31:
Related to #17744

KarlK90, 2024-03-08:
This should be solved with #22839 being merged to `master`?

jaygreco, 2024-03-09:
Whoa, thanks @KarlK90! #22839 is awesome.

The original issue is resolved, but is one step away from working turnkey:
```
$ qmk compile -kb nullbitsco/nibble -km default -e CONVERT_TO=bit_c_pro
...
Compiling: keyboards/nullbitsco/nibble/matrix.c                                                    In file included from ./lib/chibios/os/hal/include/hal_sio.h:158,
                 from ./lib/chibios/os/hal/include/hal.h:321,
                 from platforms/chibios/platform_deps.h:18,
                 from quantum/quantum.h:18,
                 from keyboards/nullbitsco/nibble/matrix.c:16:
./lib/chibios/os/hal/ports/RP/LLD/UARTv1/hal_sio_lld.h:82:2: error: #error "SIO driver activated but no UART peripheral assigned"
```
I confirmed changing `#define RP_SIO_USE_UART0` to `TRUE` in `platforms/chibios/boards/QMK_PM2040/configs/mcuconf.h` gets past it. Any thoughts?

lucasmaffazioli, 2024-04-19:
Weirdly enough I'm trying to build the firmware for the 0xCB-Helios and I'm getting the same error message:

qmk compile -kb nullbitsco/nibble -km lucasmaffazioli -e CONVERT_TO=helios

```
Compiling: keyboards/nullbitsco/nibble/matrix.c
                           In file included from ./lib/chibios/os/hal/include/hal_sio.h:158,
                 from ./lib/chibios/os/hal/include/hal.h:321,
                 from platforms/chibios/platform_deps.h:18,
                 from quantum/quantum.h:18,
                 from keyboards/nullbitsco/nibble/matrix.c:16:
./lib/chibios/os/hal/ports/RP/LLD/UARTv1/hal_sio_lld.h:82:2: error: #error "SIO driver activated but no UART peripheral assigned"
   82 | #error "SIO driver activated but no UART peripheral assigned"
      |  ^~~~~
 [ERRORS]
 |
 |
 |
make: *** [builddefs/common_rules.mk:373: .build/obj_nullbitsco_nibble_lucasmaffazioli/matrix.o] Error 1
```