qmk/qmk_firmware #22737: Outdated hint regards WSL+USB

pustekuchen91, 2023-12-23:
This hint is actually not correct. 
https://github.com/qmk/qmk_firmware/blob/master/util/qmk_install.sh#L48C3-L48C4

Actually its possible to route a usb device into the wsl (windows 11 required)
https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/

pustekuchen91, 2023-12-23:
Here is a proof that the firmware can be flashed in a devcontainer + WSL

> vscode ➜ /workspaces/qmk_firmware (add-dev-container) $ qmk flash keychron_q6_iso_encoder_keychron.bin 
> Flashing binary firmware...
> Please reset your keyboard into bootloader mode now!
> Press Ctrl-C to exit.
> 
> dfu-util 0.9
> 
> Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
> Copyright 2010-2016 Tormod Volden and Stefan Schmidt
> This program is Free Software and has ABSOLUTELY NO WARRANTY
> Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
> 
> Opening DFU capable USB device...
> ID 0483:df11
> Run-time device DFU version 011a
> Claiming USB DFU Interface...
> Setting Alternate Setting #0 ...
> Determining device status: state = dfuERROR, status = 10
> dfuERROR, clearing status
> Determining device status: state = dfuIDLE, status = 0
> dfuIDLE, continuing
> DFU mode device DFU version 011a
> Device returned transfer size 2048
> DfuSe interface name: "Internal Flash  "
> Downloading to address = 0x08000000, size = 34484
> Download        [=========================] 100%        34484 bytes
> Download done.
> File downloaded successfully
> Transitioning to dfuMANIFEST state

drashna, 2024-02-07:
> Windows 11 (Build 22000 or later)

closed 2024-06-07 by github-actions[bot]