qmk/qmk_firmware #22758: Keychron k13 support request

takinabradley, 2023-12-27:
### Issue Description

This issue is a request to support Keychron K Pro series keyboards, specifically the K13 Pro, if there hasn't been a chance to merge them yet. 

I know the the issue template asks us to refrain from keyboard requests, but the [QMK Configurator Docs](https://docs.qmk.fm/#/configurator_step_by_step?id=step-1-select-your-keyboard) suggest filing an issue if a keyboard is advertised as having QMK support. Sorry if this is an issue!

fauxpark, 2023-12-27:
https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground

closed 2023-12-27 by fauxpark

PeterMortensen, 2024-06-22:
Related: [The current software state](https://github.com/Keychron/qmk_firmware/issues/217) (the Keychron fork of QMK. etc.)

PeterMortensen, 2024-06-22:
The title is misleading (even though there isn't a K13 (like [K12](https://www.keychron.com/products/keychron-k12-wireless-mechanical-keyboard) (not based on QMK) vs. [K12 Pro](https://www.keychron.com/products/keychron-k12-pro-alice-layout-qmk-via-wireless-custom-mechanical-keyboard) (based on QMK))).