qmk/qmk_firmware #22773: [Bug]  `qmk setup` fails on Fedora 39 due to missing package 'dfu-programmer'

jbergler, 2023-12-28:
### Describe the Bug

Executing the main install command from `util/install/fedora.sh` 
```
% sudo dnf install \
        clang diffutils git gcc glibc-headers kernel-devel kernel-headers \
        make unzip wget zip python3 avr-binutils avr-gcc avr-gcc-c++ avr-libc \
        arm-none-eabi-binutils-cs arm-none-eabi-gcc-cs arm-none-eabi-gcc-cs-c++ \
        arm-none-eabi-newlib avrdude dfu-programmer dfu-util hidapi

Last metadata expiration check: 0:15:42 ago on Thu 28 Dec 2023 20:02:28.
Package diffutils-3.10-3.fc39.x86_64 is already installed.
Package git-2.43.0-1.fc39.x86_64 is already installed.
Package gcc-13.2.1-6.fc39.x86_64 is already installed.
Package glibc-devel-2.38-14.fc39.x86_64 is already installed.
Package kernel-devel-6.6.6-200.fc39.x86_64 is already installed.
Package kernel-devel-6.6.7-200.fc39.x86_64 is already installed.
Package kernel-devel-6.6.8-200.fc39.x86_64 is already installed.
Package kernel-headers-6.6.3-200.fc39.x86_64 is already installed.
Package make-1:4.4.1-2.fc39.x86_64 is already installed.
Package unzip-6.0-62.fc39.x86_64 is already installed.
Package wget-1.21.3-7.fc39.x86_64 is already installed.
Package zip-3.0-39.fc39.x86_64 is already installed.
Package python3-3.12.1-2.fc39.x86_64 is already installed.
Package avr-binutils-1:2.40-3.fc39.x86_64 is already installed.
Package avr-gcc-1:13.1.0-2.fc39.x86_64 is already installed.
Package arm-none-eabi-binutils-cs-1:2.40-2.fc39.x86_64 is already installed.
Package arm-none-eabi-gcc-cs-1:13.1.0-2.fc39.x86_64 is already installed.
No match for argument: dfu-programmer
Error: Unable to find a match: dfu-programmer
```

I can't find this package in Fedora 39.

As a workaround, removing `dfu-programmer` from the list and re-running `qmk setup` at least gets things to a stage where we can compile.

### Operating System

Fedora 39

### qmk doctor Output

```text
~/co/qmk_firmware % qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/jbergler/co/qmk_firmware
Ψ Detected Linux (Fedora Linux 39 (Workstation Edition)).
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/jbergler/co/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/jbergler/co/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/jbergler/co/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/jbergler/co/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/jbergler/co/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'wb32-dfu' boards. Run 'sudo cp /home/jbergler/co/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/jbergler/co/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/jbergler/co/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/jbergler/co/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/jbergler/co/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/jbergler/co/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/jbergler/co/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Userspace enabled: False
Ψ Git branch: main
Ψ Repo version: 0.23.2
Ψ - Latest main: 2023-12-28 12:37:01 +0800 (420b13468b) -- [Keyboard] Add meetlab kafkasplit (#22756)
Ψ - Latest upstream/master: 2023-12-28 12:37:01 +0800 (420b13468b) -- [Keyboard] Add meetlab kafkasplit (#22756)
Ψ - Latest upstream/develop: 2023-12-28 04:37:34 +0000 (73f962766e) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2023-12-28 12:37:01 +0800 (420b13468b) -- [Keyboard] Add meetlab kafkasplit (#22756)
Ψ - Common ancestor with upstream/develop: 2023-12-28 12:37:01 +0800 (420b13468b) -- [Keyboard] Add meetlab kafkasplit (#22756)
☒ Can't find dfu-programmer in your path.
[... 63 lines cut ...]
☒ Can't find dfu-programmer in your path.
Ψ Found arm-none-eabi-gcc version 13.1.0
Ψ Found avr-gcc version 13.1.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 6.4
⚠ Unknown version for dfu-programmer
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ Major problems detected, please fix these problems before proceeding.
Ψ Check out the FAQ (https://docs.qmk.fm/#/faq_build) or join the QMK Discord (https://discord.gg/Uq7gcHh) for help.
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

remembertoremind, 2024-01-06:
I also have issues with Debian, CentOS and Arch Linux. Seems like the whole `qmk setup` is borked.

fauxpark, 2024-01-06:
This is something you should take up with the respective distro packagers; it's not a QMK issue and there's not much we can do about it if they don't provide the package.

dfu-programmer is still there on Arch: https://archlinux.org/packages/extra/x86_64/dfu-programmer/
And on Debian: https://packages.debian.org/bookworm/dfu-programmer

The last release for Fedora was for 38: https://packages.fedoraproject.org/pkgs/dfu-programmer/dfu-programmer/
it seems like it has been marked as orphan, so will need someone to pick it up.

closed 2024-01-06 by fauxpark

sigprof, 2024-01-06:
One option here is to make the installation of `dfu-programmer` optional (similar to what we do for the RISC-V toolchain on Debian-like distros), so that `qmk setup` at least does something reasonable on Fedora 39; `qmk doctor` would still complain though.

In theory it could be possible to use `avrdude -c flip1` instead of `dfu-programmer`, but apparently that implementation does not have an equivalent for `dfu-programmer MCU reset` or `dfu-programmer MCU start`, therefore it may be inconvenient to use unless the actual bootloader implementation is `qmk-dfu` (which supports using a keypress to exit the bootloader).

remembertoremind, 2024-01-07:
> dfu-programmer is still there on Arch: https://archlinux.org/packages/extra/x86_64/dfu-programmer/ And on Debian: https://packages.debian.org/bookworm/dfu-programmer

You are right. I didn't have time to write a more extensive report and didn't think it would fit in this issue. I'll look into it tomorrow and create a new issue.

jonochurch, 2024-05-14:
Just came across this bug while attempting to install on Fedora 40 which came out a couple of weeks ago - I checked the install file at util/install/fedora.sh and found there is a workaround added to change the install for Fedora 39 but doesn't work for F40. Changing it to "$VERSION_ID"== "40" made it work for me

`_qmk_install() {
    echo "Installing dependencies"

    . /etc/os-release
    if [ "$VERSION_ID" == "39" ]; then
        sudo dnf $SKIP_PROMPT copr enable erovia/dfu-programmer
    fi

    # TODO: Check whether devel/headers packages are really needed
    sudo dnf $SKIP_PROMPT install \
        clang diffutils git gcc glibc-headers kernel-devel kernel-headers \
        make unzip wget zip python3 avr-binutils avr-gcc avr-gcc-c++ avr-libc \
        arm-none-eabi-binutils-cs arm-none-eabi-gcc-cs arm-none-eabi-gcc-cs-c++ \
        arm-none-eabi-newlib avrdude dfu-programmer dfu-util hidapi

    # Handle discrepancies between different Fedora versions
    sudo dnf $SKIP_PROMPT install libusb-devel \
        || sudo dnf $SKIP_PROMPT install libusb1-devel libusb-compat-0.1-devel \
        || sudo dnf $SKIP_PROMPT install libusb0-devel

    python3 -m pip install --user -r $QMK_FIRMWARE_DIR/requirements.txt
}`

BrunoSky802, 2026-02-07:
I tried changing the `$VERSION_ID` in the `util/install/fedora.sh` to the current one I'm running (43) and going thru `qmk setup` again. This did not help.

Simply running the commands from `util/install/fedora.sh` in a terminal and going thru `qmk setup` again worked:

    # Didn't use this.
    sudo dnf copr -h >/dev/null 2>&1 || sudo dnf install dnf-plugins-core
    
    # Used this, was needed.
    sudo dnf copr enable erovia/dfu-programmer
    
    # Used this, was needed.
    sudo dnf install \
         clang diffutils git gcc glibc-headers kernel-devel kernel-headers \
         make unzip wget zip python3 avr-binutils avr-gcc avr-gcc-c++ avr-libc \
         arm-none-eabi-binutils-cs arm-none-eabi-gcc-cs arm-none-eabi-gcc-cs-c++ \
         arm-none-eabi-newlib avrdude dfu-programmer dfu-util hidapi
    
    # Used this, was not needed.
    sudo dnf install libusb-devel \
        || sudo dnf install libusb1-devel libusb-compat-0.1-devel \
        || sudo dnf install libusb0-devel
    
    # The pip module for Python 3 was not included by default on my Fedora 43.
    sudo dnf install python3-pip
    
    # Used this, was needed.
    python3 -m pip install --user -r PATH_TO_qmk_firmware/requirements.txt
    
    # Used this, unsure if needed.
    qmk setup

Maybe this could be added similarly to the `Note on Debian, Ubuntu and their derivatives`, because after searching in the related issues on GitHub and a few reddit posts I almost tried the docker route, which seems to have its own set of problems.

PR #22945 (2024-01-23): Workaround for dfu-programmer on Fedora 39

## Description

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #22924
* Fixes #22773

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** document](https://docs.qmk.fm/#/pr_checklist)  [... cut]