qmk/qmk_firmware #22778: [Bug] STM32 based keyboard does not wake up propperly

tbowmo, 2023-12-29:
### Describe the Bug

Using a keychron Q11 (STM32 based), with latest develop (updated yesterday). The keyboard is connected directly to my laptop with USB. 
When I power on the laptop the keyboard is unresponsive until I unplug it momentarily. I see the same on ubuntu 22.04 and windows 11.

### Keyboard Used

keychron/q11/iso_encoder

### Operating System

Ubuntu 22.04 and windows 11

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/thomas/qmk/qmk_firmware
Ψ Detected Linux (Ubuntu 22.04.3 LTS).
Ψ Testing userspace candidate: /home/thomas/qmk/qmk_userspace -- Valid `qmk.json`
Ψ QMK userspace: /home/thomas/qmk/qmk_userspace
Ψ Userspace enabled: True
Ψ Git branch: develop
Ψ Repo version: 0.23.2
Ψ - Latest develop: 2023-12-26 07:58:21 +0800 (592a2d26ce) -- Default folder correction for rookiebwoy (#22753)
Ψ - Latest upstream/master: 2023-12-06 17:39:59 +0000 (834fb0b1fe) -- [Keyboard] Add Argyle (#22607)
Ψ - Latest upstream/develop: 2023-12-28 17:07:40 +0000 (9e50faedae) -- Remove incorrect use of WS2812_PIO_USE_PIO1 (#22771)
Ψ - Common ancestor with upstream/master: 2023-12-06 17:39:59 +0000 (834fb0b1fe) -- [Keyboard] Add Argyle (#22607)
Ψ - Common ancestor with upstream/develop: 2023-12-26 07:58:21 +0800 (592a2d26ce) -- Default folder correction for rookiebwoy (#22753)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f90)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

KarlK90, 2023-12-30:
@tbowmo please try my PR #21656 if this fixes the issues you are seeing.

tbowmo, 2023-12-30:
@KarlK90 Tried your PR, but doesn't make any change. Keyboard is still dead until i unplug / plug it, after powering on the pc.

tbowmo, 2023-12-30:
Further debugging, it seems that it only happens when doing a cold boot. If I just reboot the computer the keyboard enumerates correctly on the USB port. But if the pc has been powered off, the keyboard does not enumerate until it's re-plugged.

Just a wild guess but can the usb driver timeout during poweron, and "crash" if there is no host to enumerate it?

sigprof, 2023-12-30:
Keychron Q11 is a split board without a VBUS detection pin, therefore it probably needs the split watchdog to handle that case — try adding this to `config.h`:
```c
#define SPLIT_WATCHDOG_ENABLE
```

tbowmo, 2023-12-30:
@sigprof even though that it doesn't make sense (reading the docs about SPLIT_WATCHDOG_ENABLE) it actually fixed the issue.

According to the docs, it resets the _slave_ but in my case it's the _master_ that hangs (as it is the master that is connected to the host, and does not enumerate correctly)

sigprof, 2023-12-31:
The problem is that without a VBUS detection pin the master side detection does not work if there is a large delay between supplying the power and actually enumerating the USB device, and that kind of delay often happens when the computer gets powered on from the fully off state.  Then both sides start up in the slave mode, and without `SPLIT_WATCHDOG_ENABLE` the master detection is never retried — both sides turn off USB and just wait for the serial communication from the master side.

Currently QMK does not have any way to switch between the master and slave roles at runtime — the choice needs to be made during the firmware initialization; the `SPLIT_WATCHDOG_ENABLE` code works around that problem by initiating a full restart of the firmware, so that the initial master side detection could be retried.

tbowmo, 2023-12-31:
@sigprof ok, it makes more sense now.

So this actually needs to be changed in the qmk repo as well for this keyboard, as it's not just a keymap issue. I'll create a pr "next year" ;)

closed 2024-01-02 by tbowmo