qmk/qmk_firmware #22790: [Bug] Error when flashing skyloong gk61 pro: [platforms/chibios/flash.mk:98: flash] Error 74

blaiddx64, 2023-12-31:
### Describe the Bug

When I try to flash default km to gk61 pro using qmk compile/qmk flash it returns no error, but the keyboard does not work, only the RGB leds works (it was working before flash).

If I follow the commands on it's [page](https://github.com/qmk/qmk_firmware/tree/master/keyboards/skyloong/gk61/pro) (make), it returns error 74:

```bash
~/qmk_firmware (master)> sudo make skyloong/gk61/pro:default
QMK Firmware 0.23.2
Making skyloong/gk61/pro with keymap default

arm-none-eabi-gcc (Arch Repository) 13.2.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  39546	      0	  39546	   9a7a	skyloong_gk61_pro_default.bin

Compiling: quantum/command.c                                                                        [OK]
Linking: .build/skyloong_gk61_pro_default.elf                                                       [OK]
Creating binary load file for flashing: .build/skyloong_gk61_pro_default.bin                        [OK]
Creating load file for flashing: .build/skyloong_gk61_pro_default.hex                               [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  39546	      0	  39546	   9a7a	skyloong_gk61_pro_default.bin

Copying skyloong_gk61_pro_default.bin to qmk_firmware folder
```

```bash
~/qmk_firmware (master)> sudo make skyloong/gk61/pro:default:flash
QMK Firmware 0.23.2
Making skyloong/gk61/pro with keymap default and target flash

arm-none-eabi-gcc (Arch Repository) 13.2.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  39546	      0	  39546	   9a7a	skyloong_gk61_pro_default.bin

Compiling: quantum/command.c                                                                        [OK]
Linking: .build/skyloong_gk61_pro_default.elf                                                       [OK]
Creating binary load file for flashing: .build/skyloong_gk61_pro_default.bin                        [OK]
Creating load file for flashing: .build/skyloong_gk61_pro_default.hex                               [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  39546	      0	  39546	   9a7a	skyloong_gk61_pro_default.bin

Copying skyloong_gk61_pro_default.bin to qmk_firmware folder                                        [OK]
Flashing for bootloader: stm32duino
Bootloader not found. Make sure the board is in bootloader mode. See https://docs.qmk.fm/#/newbs_flashing
 Trying again every 0.5s (Ctrl+C to cancel).................
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
Device ID 1eaf:0003
Device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Interface #2 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download	[=========================] 100%        39548 bytes
Download done.
DFU state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Resetting USB to switch back to runtime mode
Done!
make[1]: *** [platforms/chibios/flash.mk:98: flash] Error 74
Make finished with errors
make: *** [Makefile:416: skyloong/gk61/pro:default:flash] Error 1
```

### Keyboard Used

skyloong/gk61/pro

### Operating System

Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/blaidd/qmk_firmware
Ψ Detected Linux (EndeavourOS).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.23.2
Ψ - Latest master: 2023-12-29 17:17:20 +0100 (fc20e0b014) -- Waffling60 iso rev e (#22733)
Ψ - Latest upstream/master: 2023-12-29 17:17:20 +0100 (fc20e0b014) -- Waffling60 iso rev e (#22733)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-12-29 17:17:20 +0100 (fc20e0b014) -- Waffling60 iso rev e (#22733)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.2.0
Ψ Found avr-gcc version 13.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.0.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

keyboard-magpie, 2023-12-31:
Are you sure you have a gk61 pro?

blaiddx64, 2023-12-31:
> Are you sure you have a gk61 pro?

100%

sigprof, 2023-12-31:
There is some chance that you actually have the `skyloong/gk61/pro_48` PCB.  Apparently the `pro` and `pro_48` versions can be distinguished by looking at the MCU chip package — it's 64-pin for the `pro` version, and 48-pin for `pro_48`: https://discord.com/channels/440868230475677696/473506116718952450/1185343773341917244

blaiddx64, 2023-12-31:
> There is some chance that you actually have the `skyloong/gk61/pro_48` PCB. Apparently the `pro` and `pro_48` versions can be distinguished by looking at the MCU chip package — it's 64-pin for the [...]

I didn't pay attention that there's 2 gk61 pro versions, it solved the problem. Thank you so much!

closed 2023-12-31 by blaiddx64