qmk/qmk_firmware #22818: Feker IK65 not in the list of qmk.fm

alexanderjacques, 2024-01-03:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hello,

The Feker IK65 is VIA compatible but it's not in the list of keyboards yet of qmk.fm.
Anyone who can add in the list of keyboards? 

Don't hesitate to reach out to me for more info.

fauxpark, 2024-01-03:
VIA compatibility does not necessarily imply QMK compatibility.

alexanderjacques, 2024-01-03:
> VIA compatibility does not necessarily imply QMK compatibility. This board is wireless, so it is unlikely to have an MCU that is supported.

It has following chip "VS12L17A"  not sure if that's helpful or not

zvecr, 2024-01-05:
From the top of the qmk.fm/keyboards page,

> We used to host/update the binaries for each keyboard here, but the tooling we used for this is no longer available to us, so both this list and the binaries are out of date.

But also mentioned in the issue template...

> Please refrain from asking maintainers to add support for specific keyboards -- it is unlikely they will have hardware available, and will not be able to help.
Your best bet is to take the initiative, add support, then submit a PR yourself.

closed 2024-01-05 by zvecr