qmk/qmk_firmware #22820: Issues with Keychron Q3 Pro (ISO Knob Version) Firmware, Bluetooth Connectivity, and Configuration Software

shm0rt, 2024-01-03:
### Issue Description

## Problem:
After downloading the firmware from the official website for my Keychron Q3 Pro (ISO Knob version), I've encountered several issues that are affecting the functionality of the keyboard. Initially, the keyboard displayed erratic behavior. While downloading the SE binary seemed to mitigate some problems, the Bluetooth connectivity remains non-functional.

## Additional Issue:
Moreover, the software provided for keyboard configuration is not operating correctly. Despite using a wired connection, which is my usual preference, and attempting to configure the keyboard mappings through the Brave browser on Windows (with all security settings for the site disabled and the browser updated to the newest version), the software fails to recognize the keyboard or allow any mapping changes.

## Impact:
These issues are particularly concerning given the substantial investment made into this keyboard, amounting to several hundred euros. The expected functionality and support for a device of this caliber are not met, leading to a disappointing user experience.

## Request:
Could you please provide assistance, updates, or workarounds to address these problems? Any guidance on how to resolve the Bluetooth connectivity and software configuration issues would be greatly appreciated.

Thank you for your attention to this matter.

keyboard-magpie, 2024-01-03:
This keyboard is not part of the qmk_firmware repository- this issue should be re-posted on the keychron fork where it is hosted: https://github.com/Keychron/qmk_firmware/tree/playground

closed 2024-01-03 by keyboard-magpie

shm0rt, 2024-01-04:
Issue moved on keychron fork: https://github.com/Keychron/qmk_firmware/issues/210