qmk/qmk_firmware #22833: [Bug] * LED_MATRIX_DRIVER=CKLED2001 is not a valid matrix type

Dialgatrainer02, 2024-01-05:
### Describe the Bug

for context ive never programed in c or used git before
im trying tp update the qmk version to latest from https://github.com/mattmasc/qmk_firmware/tree/bluetooth_playground and it has merge conflicts i have tried to solve in my fork however it wont complie and gives me an error 

 * RGB_MATRIX_DRIVER=CKLED2001 is not a valid matrix type                                              [ERRORS]

builddefs/common_features.mk:452: *** Invalid RGB_MATRIX_DRIVER.  Stop.
make: *** [Makefile:416: keychron/k8_pro/ansi/rgb:via] Error 1

im unsure where the error is as i can successfully compile on the old version but not the new 

steps to reporoduce
fork qmk 
add remote https://github.com/mattmasc/qmk_firmware/tree/bluetooth_playground
cherry pick commits from keychron 
deal with merge conflicts
attempt to complie
  

### Keyboard Used

keychron/k8_pro

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-k8-pro-qmk-via-wireless-mechanical-keyboard

### Operating System

arch linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/olivia/qmk_firmware
Ψ Detected Linux (Arch Linux).
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/olivia/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/olivia/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/olivia/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/olivia/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/olivia/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'wb32-dfu' boards. Run 'sudo cp /home/olivia/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/olivia/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/olivia/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/olivia/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/olivia/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/olivia/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/olivia/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Userspace enabled: False
Ψ Git branch: bluetooth_playground
Ψ Repo version: 0.23.3
Ψ - Latest bluetooth_playground: 2023-05-01 02:08:47 +0200 (3205716774) -- Cherry-Pick Keychron's bluetooth_playground changes
Ψ - Latest upstream/master: 2024-01-04 03:59:38 -0500 (b02abcb9a6) -- Regenerate Files (#22821)
Ψ - Latest upstream/develop: 2024-01-05 04:27:55 -0800 (7ed3138e5f) -- Remove PWM advanced check for WS2812 driver (#22830)
Ψ - Common ancestor with upstream/master: 2024-01-04 03:59:38 -0500 (b02abcb9a6) -- Regenerate Files (#22821)
Ψ - Common ancestor with upstream/develop: 2024-01-04 03:59:38 -0500 (b02abcb9a6) -- Regenerate Files (#22821)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.2.0
Ψ Found avr-gcc version 13.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.0.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb16)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb3)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

i originally tried to port to vial and to test if vial wasn't working i tried to do basic qmk which is still failing
 should i also contact keychron if its unsupported here

fauxpark, 2024-01-06:
`CKLED2001` has been renamed to `snled27351`, and all driver names are now lowercase.

closed 2024-01-06 by fauxpark

Dialgatrainer02, 2024-01-06:
where does a keyboard define which driver it uses as im curently trying to change it as i reapplied the cherry pick keeping upstream for any rgb and led  matrix files but the keyboard still wants to use the old driver name

PeterMortensen, 2025-01-14:
CKLED2001 was [renamed to SNLED27351](https://www.reddit.com/r/Keychron/comments/1dk4s03/comment/l9h51e8/) ([numero 22071](https://github.com/qmk/qmk_firmware/pull/22071). 2023-10-04. Mentioned in [the 2023-11 change log](https://github.com/qmk/qmk_firmware/tree/master/docs/ChangeLog/20231126.md), near "#22071")