qmk/qmk_firmware #22850: [Feature Request] Allow forcing individual community layouts in `qmk.json` for `userspace-compile`  and GitHub Actions

bcat, 2024-01-07:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

See [QMK Discord discussion](https://discord.com/channels/440868230475677696/1146201809224994887/1193407942616174744) for additional context.

The userspace build infra used by `qmk userspace-compile` and the default GitHub Actions setup seems to treat each keyboard to compile as a pair of keyboard name and keymap name. This leads to problems when one has multiple keyboards built with the same PCB but different community layouts.

For example, I have a `dz60` PCB with the `60_ansi_split_bs_rshift` layout and one with the `60_tsangan_hhkb`. My current [handwritten build script](https://github.com/bcat/qmk_userspace/blob/main/users/bcat/compile.sh) compiles both by passing the `FORCE_LAYOUT` environment variable to `make`. I'd like to start using `qmk userspace-compile` instead, but `qmk userspace-add` only lets me set up a `qmk.json` file like this:

```
{
    "userspace_version": "1.0",
    "build_targets": [
        ["dz60", "bcat"]
    ]
}
```

There doesn't seem to be a way to build the `dz60` with a particular community layout, nor to build it twice with different community layouts. I would suggest an amended `qmk.json` format like this instead that allowed specifying a particular `FORCE_LAYOUT` value for each firmware file to build:

```
```
{
    "userspace_version": "1.0",
    "build_targets": [
        {"kb": "dz60", "km": "bcat", "layout": "60_ansi_split_bs_rshift"},
        {"kb": "dz60", "km": "bcat", "layout": "60_tsangan_hhkb"}
    ]
}
```

bcat, 2024-01-07:
Note that I think this is distinct from https://github.com/qmk/qmk_firmware/issues/22815. That issue seems to be about the existing ability to pass force layout to `qmk compile` not working, but even if that bug was fixed, it still wouldn't address this FR. (Passing `-e FORCE_LAYOUT=foo` to `qmk userspace-compile` would try to force the same community layout for every keyboard, which is probably not what anybody wants.)

tzarc, 2024-01-08:
I think it'd be more likely we'd end up with per-keyboard environment variables added when invoking `qmk userspace-add`, with the usual `-e FORCE_LAYOUT=xxx` at that point -- resulting in something like:
```json
{
    "userspace_version": "1.1",
    "build_targets": [
        [
            "dz60",
            "bcat",
            [
                ["FORCE_LAYOUT", "60_ansi_split_bs_rshift"]
            ]
        ],
        [
            "dz60",
            "bcat",
            [
                ["FORCE_LAYOUT", "60_tsangan_hhkb"]
            ]
        ],
        ["40percentclub/luddite", "default"],
        [
            "40percentclub/luddite",
            "default",
            [
                ["FORCE_LAYOUT", "60_ansi"],
                ["CONVERT_TO", "proton_c"]
            ]
        ]
    ]
}
```

bcat, 2024-01-09:
Ah, yeah, that would address my use case, and it's more flexible too.

tzarc, 2024-01-11:
It'd be great if you could give #22887 a go to see if it addresses your issue.

bcat, 2024-04-27:
Not stale, and has a fix in flight. :)

bcat, 2024-07-27:
Still not stale, and I can confirm PR 22887 seems to work.

PR #22887 (2024-08-12): Userspace: add support for adding environment variables during build
## Description

A few complaints on Discord and the like for not being able to specify things like `CONVERT_TO` and `FORCE_LAYOUT` when using userspace.

This adds support for the usual `-e`/`--env` parameters to `qmk userspace-add` and `qmk userspace-remove`, and ensures such extra environment variables are propagated across to each build.

Userspace repo version is bumped to `1.1` as a result of the addition to the file format.

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
-  [... cut]
closed 2024-08-12 by tzarc