qmk/qmk_firmware #22916: [Bug] Retro tapping not triggered after release of dual function key during key roll over from previous keys

ErikSchlyter, 2024-01-18:
### Describe the Bug

When a dual function key is configured as a normal key when _pressed_, but to enter a layer when _hold_; the _retro tapping_ feature doesn't trigger the normal key if an arbitrary key from previous typing is still down (key roll over).

I have recently acquired a ZSA Voyager and I have configured a thumb key to send `Enter` when pressed, and to enable my navigation layer when hold. Since I want quick access to the nav layer while typing, I have configured _tapping term_ to a minimum (1ms), and enabled _retro tapping_. This means I can type fairly quickly and still be able to access my arrow keys directly in the middle of a sentence. Thanks to _retro tapping_, I can still use the thumb key as the very common `Enter` key. It works great!

However; if I'm currently holding a previous key while pressing down the thumb key, the _retro tapping_ functionality isn't triggering. The keyboard isn't sending `Enter` once I release the dual function key (without pressing anything in the nav layer).

Steps to reproduce:

1. Configure _tapping term_ to 1ms
2. Enable _retro tapping_
3. Configure a dual function key; a normal character when _pressed_ (e.g. `A`), and a specific layer when _hold_.

Do the following in a short time span of about a second or so:
1. Press any other arbitrary normal key, e.g. `B`
2. Press the dual function key
3. Release the `B` key
4. Release the dual function key

Expected behavior at each step:
1. The keyboard should send that `B` is pressed
2. The keyboard should enter the layer (basically directly; 1ms)
3. The keyboard should send that `B` is released.
4. The keyboard should exit the layer, and then send `A`, since nothing in the layer was pressed and _retro tapping_ is enabled.

Actual observed behavior:
1. The keyboard sends that `B` is pressed.
2. The keyboard enters the layer (this is visible, because the LEDs light up)
3. The keyboard sends that `B` is released.
4. The keyboard exits the layer (the LEDs go dark), but the `A` key is not sent.

From what I understand this is a rather unusual configuration, i.e. having _tapping term_ set to 1ms, but it is quite crucial to the way I want to operate my keyboard. I have set my thumb key to access the arrow keys because I want navigation available without any delay at all, and that part works great. I still need to repurpose the thumb keys for `Enter`/`Space` (not enough thumb keys, ah the misery!), but the observed bug causes strife in my daily use, because when executing some arbitrary short shell script commands (e.g. `ls`) I basically roll over all three keys `L`, `S`, `Enter`. This happens quite often, and those bash commands just sit there and stare at me menacingly from the terminal because `Enter` never arrives.

I know ZSA maintains their own fork of QMK, but I suspect the problem stems from the original repo. I haven't checked out the code myself yet, but if anyone else wants to have a go at this, you have my undying gratitude :)

### Keyboard Used

voyager

### Link to product page (if applicable)

https://www.zsa.io/voyager

### Operating System

Linux (Manjaro)

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

amatoenot, 2024-02-15:
@ErikSchlyter It seems like for your specific case, it's better to enable the 'Hold On Other Key Press' setting to be able to select the HOLD action on other key press without the delay. Instead of retro tap, just set TAPPING_TERM to something regular like 200ms (or higher to mimic retro tap behavior). This will allow you to instantly switch to the arrow keys layer on hold and rollover the Enter tap key without any issues.

I have a case where I want to have a relatively quick hold decision timeout for a LT key (tapping term 150 ms) but at the same time, I need the tap key to be triggered after the tapping term has passed (using retro tap). And I noticed the same issue, that retro tap just doesn't work on rollover (with LT with retro tap being the last one). So it definitely still looks like an issue.

PeterBul, 2024-04-09:
I am having the same issue. Holding and releasing a dual function key doesn't send the tap stroke even though retro is enabled when the hold function is a MO layer shift. I tried the "Hold on other key press" approach, but that doesn't work with with the layer shift either. There the base layer stroke is still chosen even though I press the layer shift key and another key. I still have to wait for the layer to switch to get the right stroke.

JohnRigoni, 2024-04-26:
Hello, I have a fix for this behavior here https://github.com/JohnRigoni/qmk_firmware. I am going to do some further testing/tweaking then make a pull request.

JohnRigoni, 2025-02-16:
It's merged 🎉