qmk/qmk_firmware #22993: Inverting the pin "click" state for pointing device.

fire1, 2024-01-31:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

HI!

I moded my Splitkb Aurora v3, right side with a pointer device from Thumbstick for Oculus Quest 2 (since this can fit perfectly in the pace of the Rotary encoder).

Everything is working fine with the axis when I added:
```
void pointing_device_init_user(void) {
    setPinInputHigh(ANALOG_JOYSTICK_X_AXIS_PIN);
    setPinInputHigh(ANALOG_JOYSTICK_Y_AXIS_PIN);
}
```

 but the button click is not registered since the joystick sends a HIGH state instead of LOW.
Somehow the "Click pin" is always in the "HIGH" state,
 I tried to set  `if (!is_keyboard_master()) setPinInput(ANALOG_JOYSTICK_CLICK_PIN);`, but the pin state is not changed.

Note: 
_Before I made this hardware mode I tested the pin arrangement for the joystick, flipping the VCC and GND for the joystick will fix the "click" button, but the axis (left/down) are lost (right and up are fine)._

drashna, 2024-02-07:
This line is the issue: 
https://github.com/qmk/qmk_firmware/blob/master/drivers/sensors/analog_joystick.c#L83-L85

And that would need to be fixed/changed.  

Eg, something like: 
```c
#ifdef ANALOG_JOYSTICK_CLICK_PIN
#    if ANALOG_JOYSTICK_CLICK_PIN_ON_STATE == 1
    report.button = !readPin(ANALOG_JOYSTICK_CLICK_PIN);
#    else 
    report.button = readPin(ANALOG_JOYSTICK_CLICK_PIN);
#    endif
#endif

fire1, 2024-02-12:
> This line is the issue: https://github.com/qmk/qmk_firmware/blob/master/drivers/sensors/analog_joystick.c#L83-L85
>
> And that would need to be fixed/changed.
>
> Eg, something like:
>
> ```c
> #ifdef [...]

Exactly, the fix is easy.
Thanks for attention...

closed 2024-06-12 by github-actions[bot]