qmk/qmk_firmware #23036: [Bug] qmk setup command should not require a yes answer to cloning the qmk_firmware repo to install dependencies

benjistokman, 2024-02-08:
### Describe the Bug

`qmk setup` should not require a yes answer to cloning the qmk_firmware repo to install dependencies. I only wanted to install the dependencies, not clone the repository.

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2024-02-08:
This is a chicken-and-egg problem, the dependency installation scripts live within QMK Firmware.

I'm not sure what you're asking for is achievable -- if you're not building for QMK you probably don't need the dependencies?

benjistokman, 2024-02-08:
I have the firmware on a user that doesn't have sudo access. Perhaps having a copy-paste list of commands?

fauxpark, 2024-02-08:
The install scripts are a little too lengthy for that.

tzarc, 2024-02-14:
....and for normal distributions, require `sudo` access. So you'll likely need to inspect the installation scripts, build the dependencies yourself and place them on `$PATH` inside the user account.

I'm all for keeping things self-contained, but asking users who aren't necessarily familiar with Unix-isms to build their own tooling is probably a bridge too far.