qmk/qmk_firmware #23053: [Bug] keyboards/kinesis/kint41 hangs for 10 seconds or indefinitely

stevenbetten, 2024-02-11:
### Describe the Bug

Since commit 416af0171c6433a7ecb198386dd2c3ac70d4cbd2, keyboards/kinesis/kint41 hangs for 10 seconds or indefinitely. I experienced an indefinite hang on my kint41 keyboard after flashing the result of `make kinesis/kint41:default` built from the head of qmk_firmware. In contrast, the keyboard works well when flashed from qmk_firmware tag 0.21.6. I found https://github.com/kinx-project/kint/issues/77, which describes other users experiencing the issue.

### Keyboard Used

kinesis/kint41

### Operating System

macOS 13.6.3 (22G436)

### qmk doctor Output

```text
steve@Stevens-MacBook-Pro qmk_firmware % qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /Users/steve/qmk_firmware
Ψ Detected macOS 13.6.3 (Apple Silicon).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.23.9
Ψ - Latest master: 2024-02-06 18:09:22 +1100 (45ae4dec4d) -- WS2812: Better error message when trying to use `bitbang` driver on RP2040 (#23025)
Ψ - Latest upstream/master: 2024-02-06 18:09:22 +1100 (45ae4dec4d) -- WS2812: Better error message when trying to use `bitbang` driver on RP2040 (#23025)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-02-06 18:09:22 +1100 (45ae4dec4d) -- WS2812: Better error message when trying to use `bitbang` driver on RP2040 (#23025)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

stevenbetten, 2024-02-12:
fauxpark closed the corresponding pull request a few hours ago as "This doesn't do anything." but I asked them to explain why they think that, because I disagree. See https://github.com/qmk/qmk_firmware/pull/23054.

tzarc, 2024-02-14:
Replicating my comment from the now-closed PR:

---

For clarity, the ChibiOS code for MIMXRT1062 was apparently written purely for the kint41. It seems to be a bare-bones and potentially broken implementation -- it's been reported that anything to do with timing in QMK is flat-out broken on MIMXRT1062-based boards.

I've attached two files -- disassembly from before and after applying the PR. Apart from the `WFI` call, there seems to be a missing `port_unlock()`? Perhaps there's an issue with preprocessor in the underlying ChibiOS port, mistakenly removing the wrong function call?

[kinesis_kint41_default.elf.base.dis.txt](https://github.com/qmk/qmk_firmware/files/14259338/kinesis_kint41_default.elf.base.dis.txt)
[kinesis_kint41_default.elf.pr.dis.txt](https://github.com/qmk/qmk_firmware/files/14259340/kinesis_kint41_default.elf.pr.dis.txt)

I'd say the general issue here is that the usual interrupts aren't firing, and ChibiOS isn't scheduling the QMK main thread for execution as a result. Interrupts such as USB or SysTick tend to keep QMK running, and perhaps these aren't enabled in the MIMXRT1062 port.

Also something to at least _try_ -- ChibiOS creates an idle thread even though QMK should be running as a hard loop -- you could try adding `-DCH_CFG_NO_IDLE_THREAD=TRUE` to instruct ChibiOS not to create the idle thread. This isn't likely going to be accepted as a fix; it's more we don't have any MIMXRT1062 boards that we can test with -- adding the define will tell us more about the correct course of action.

stevenbetten, 2024-02-15:
Nick, thanks for the additional comments!-Steve

robertbcalhoun, 2025-04-09:
Verified this issue still present in qmk (or ChibiOS) as of qmk a0f40d9410d1ecac4b498b999ebd04e65ef762f4.

* start with clean install of qmk on ubuntu 24.04.
* build default firmware for kinesis/kint41 with:
```
qmk compile -kb kinesis/kint41 -km default
```
* after loading firmware and rebooting, board stays in (or returns to) bootloader mode and keyboard is not usable.
* verified that adding `OPT_DEFS += -DCORTEX_ENABLE_WFI_IDLE=FALSE` to `rules.mk` as suggested by @stevenbetten and @isaacgale in https://github.com/kinx-project/kint/issues/77 works around the issue.
* verified that adding `OPT_DEFS +=-DCH_CFG_NO_IDLE_THREAD=TRUE` to `rules.mk` as suggested by @tzarc also works around the issue.

I don't know which workaround is preferred. I am running with `OPT_DEFS +=-DCH_CFG_NO_IDLE_THREAD=TRUE` as @tzarc says qmk doesn't use the idle thread anyway.

If someone new to the issue is reading this, basic keyboard functionality can be restored by flashing with the (obsolete) precompiled firmware for Teensy 4.1 available at https://github.com/kinx-project/kint/blob/main/default-firmware/kinesis_kint41_default.hex  .

rwg0, 2025-10-26:
Can also confirm that the issue with delayed boot, failure to properly exit bootloader mode without power cycle, etc still exist at 98e65cf3e1fb4bf61367d469a62d3cce0a9fac20 

The one line fix of adding OPT_DEFS += -DCORTEX_ENABLE_WFI_IDLE=FALSE to the kint41 rules.mk resolves the issues fully for me.

Firmwares created from the configurator show all of the above issues due to this change still not being in the codebase - it seems strange not to take a pragmatic approach to making this work again.