qmk/qmk_firmware #23069: [Bug] `QK_MAGIC_SWAP_LCTL_LGUI` cannot be used with `register_code(<kc>)`

aamkye, 2024-02-13:
### Describe the Bug

I'm trying to define a swap function for `QK_MAGIC_SWAP_LCTL_LGUI` and `QK_MAGIC_UNSWAP_LCTL_LGUI` as there is no toggle function for that with following code:

```c++
bool is_kc_swit_enabled = false;

enum custom_keycodes {
    KC_SWIT
};

...

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
  switch (keycode) {
    case KC_SWIT: {
      if (record->event.pressed) {
        if (is_kc_swit_enabled) {
          register_code(QK_MAGIC_SWAP_LCTL_LGUI);
        } else {
          register_code(QK_MAGIC_UNSWAP_LCTL_LGUI);
        }
        is_kc_swit_enabled = !is_kc_swit_enabled;
      }
      break;
    }
  }
  return true;
}
```

But it produces me error:

```text
> qmk compile -kb keychron/q4/ansi/v2 -km custom -j $(nproc)
Ψ Compiling keymap with gmake --jobs=16 --output-sync=target -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=keychron/q4/ansi/v2 KEYMAP=custom KEYBOARD_FILESAFE=keychron_q4_ansi_v2 TARGET=keychron_q4_ansi_v2_custom INTERMEDIATE_OUTPUT=.build/obj_keychron_q4_ansi_v2_custom VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

Size before:
   text    data     bss     dec     hex filename
      0   43540       0   43540    aa14 keychron_q4_ansi_v2_custom.bin

arm-none-eabi-gcc (Homebrew ARM GCC 8.5.0_1) 8.5.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: quantum/keymap_introspection.c                                                          In file included from quantum/keymap_introspection.c:5:
./keyboards/keychron/q4/ansi/keymaps/custom/keymap.c: In function 'process_record_user':
./keyboards/keychron/q4/ansi/keymaps/custom/keymap.c:75:25: error: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '28695' to '23' [-Werror=overflow]
           register_code(CG_LSWP);
                         ^~~~~~~
./keyboards/keychron/q4/ansi/keymaps/custom/keymap.c:77:25: error: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '28696' to '24' [-Werror=overflow]
           register_code(CG_LNRM);
                         ^~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
gmake: *** [builddefs/common_rules.mk:373: .build/obj_keychron_q4_ansi_v2_custom/quantum/keymap_introspection.o] Error 1
```

### Keyboard Used

keychron/q4/ansi/v2

### Link to product page (if applicable)

-

### Operating System

MacOS

### qmk doctor Output

```text
> qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /Users/xxx/yyy/qmk_firmware
Ψ Detected macOS 14.1.1 (Apple Silicon).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.23.9
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

-

### Additional Context

-

tzarc, 2024-02-13:
Not a bug, `register_code` et.al. register a keycode in the USB keyboard report as "pressed".
QMK-specific keycodes such as swapping ctl/gui are handled purely within the keyboard, so there's nothing to actually report over USB upon press.

Given how you're using it, you're better off placing `QK_MAGIC_SWAP_LCTL_LGUI` in your keymap directly, and if needed, maintaining the boolean's state by handling the case within `process_record_user` as you currently do. There's no need to define a new keycode for it.

tzarc, 2024-02-13:
Actually, reading things further, if we're missing a toggle swap keycode for LCTL+LGUI, then that's the real bug. `register_code` etc. aren't the issue here.

closed 2024-02-13 by tzarc

tzarc, 2024-02-13:
And for now, you could just do:

```c
                case KC_SWIT: {
                    if (record->event.pressed) {
                        keymap_config.swap_lctl_lgui = !keymap_config.swap_lctl_lgui;
                    }
                    break;
```

aamkye, 2024-02-14:
> Actually, reading things further, if we're missing a toggle swap keycode for LCTL+LGUI, then that's the real bug. `register_code` etc. aren't the issue here.
> 
> EDIT: there's `QK_MAGIC_TOGGLE_CTL_GUI`/`CG_TOGG`, but that's both sides. Does that not suffice in your case?

No, unfortunately not, but the snippet that you've provided works like a charm <3

aamkye, 2024-02-14:
@tzarc, is this part documented somewhere in the docs, or is it tribal knowledge?

```
extern keymap_config_t keymap_config;
keymap_config.swap_lctl_lgui = !keymap_config.swap_lctl_lgui;
```

tzarc, 2024-02-14:
I just looked at what the swap/unswap keycodes were doing and extrapolated, really.
Had a quick look and there are a few keymaps doing similar things in the repository already anyway.