qmk/qmk_firmware #23079: Compilation error: quantum/keymap_introspection.c

bill-alan, 2024-02-14:
Hello everyone, I am a new user of qmk for the first time who also asked questions yesterday. I have a new problem today and hope to consult you.
I followed the file and found no problem, and the previous code without RGB matrix and rotation encoder can be compiled. Usage should not be an issue for quantum.

Compiling: quantum/keymap_introspection.c                                                          In file included from quantum/keymap_introspection.c:12:
quantum/keymap_introspection.h:12:1: error: expected ',' or ';' before 'uint8_t'
   12 | uint8_t keymap_layer_count_raw(void);
      | ^~~~~~~
 [ERRORS]
 |
 |
 |
make: *** [builddefs/common_rules.mk:373：.build/obj_hwpad2_default/quantum/keymap_introspection.o] 

[image]

fauxpark, 2024-02-14:
The issue is in your keymap, this is just a side effect of how QMK compiles it.

fauxpark, 2024-02-14:
You're missing a semicolon.

bill-alan, 2024-02-14:
> You're missing a semicolon.
endif before, my question. Thank you very much for your help.

closed 2024-02-14 by fauxpark