qmk/qmk_firmware #23098: [Bug] qmk setup can't handle spaces in home path

Nobodi99, 2024-02-16:
### Describe the Bug

I was just upgrading to the external userspace and had to update my qmk config because I couldn't get the userspace-add command to work.

I then saw this line in my config:
`user.qmk_home=S:ProjectsKeyboardsQMK`

This was really weird and I tried to run `qmk setup` again but it set it to the above so I had to set it manually.
I had to set the path sorrounded with `""` to set it to:
`user.qmk_home=S:/Projects/Keyboards/QMK Firmware/qmk_firmware`

After setting it manually it finally works.
I would expect that `qmk setup` would set the path correctly even if it contains a space.
Sadly I don't know enough about the build code to do any PR for it :)

### Operating System

Windows

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2024-02-17:
QMK builds won't work on paths with spaces regardless. It's an issue with GNU Make and not something we can fix.

Nobodi99, 2024-02-17:
> QMK builds won't work on paths with spaces regardless. It's an issue with GNU Make and not something we can fix.

I can compile the firmware after setting the path like this. And I never had any problem compiling before this change to external userspaces.
I do run into the problem that it can't save the uf2 file to the userspace folder now because the userspace is in another folder with a space in it and somehow it cuts of everything before the space.
Would you happen to know which repository I could look into for that?
(The uf2 file is succesfully created in the .build folder of the main repo though)