qmk/qmk_firmware #23107: [Bug] QMK setup warning and crash on new-keyboard

JimmyBlackley, 2024-02-19:
### Describe the Bug

qmk setup completes with a warning, as outlined in the qmk doctor

I tried running brew reinstall libimagequant as outlined here https://github.com/qmk/qmk_firmware/issues/22283#issuecomment-1765381093
This did not solve

running qmk new-keyboard causes a crash:

``` james@James's MPB ~ % qmk new-keyboard
Error: %s: %s ('ImportError', ImportError("dlopen(/usr/local/Cellar/qmk/1.1.2_3/libexec/lib/python3.11/site-packages/PIL/_imaging.cpython-311-darwin.so, 0x0002): Library not loaded: '/usr/local/opt/libimagequant/lib/libimagequant.0.0.4.dylib'\n  Referenced from: '/usr/local/Cellar/qmk/1.1.2_3/libexec/lib/python3.11/site-packages/PIL/_imaging.cpython-311-darwin.so'\n  Reason: tried: '/usr/local/opt/libimagequant/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/local/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/local/Cellar/libimagequant/4.3.0/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/local/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/lib/libimagequant.0.0.4.dylib' (no such file)"))
Traceback (most recent call last):
  File "/usr/local/Cellar/qmk/1.1.2_3/libexec/lib/python3.11/site-packages/qmk_cli/script_qmk.py", line 76, in main
    import qmk.cli  # noqa
    ^^^^^^^^^^^^^^
  File "/Users/james/qmk_firmware/lib/python/qmk/cli/__init__.py", line 252, in <module>
    __import__(subcommand)
  File "/Users/james/qmk_firmware/lib/python/qmk/cli/painter/__init__.py", line 1, in <module>
    from . import convert_graphics
  File "/Users/james/qmk_firmware/lib/python/qmk/cli/painter/convert_graphics.py", line 7, in <module>
    from qmk.painter import render_header, render_source, render_license, render_bytes, valid_formats
  File "/Users/james/qmk_firmware/lib/python/qmk/painter.py", line 6, in <module>
    from PIL import Image, ImageOps
  File "/usr/local/Cellar/qmk/1.1.2_3/libexec/lib/python3.11/site-packages/PIL/Image.py", line 103, in <module>
    from . import _imaging as core
ImportError: dlopen(/usr/local/Cellar/qmk/1.1.2_3/libexec/lib/python3.11/site-packages/PIL/_imaging.cpython-311-darwin.so, 0x0002): Library not loaded: '/usr/local/opt/libimagequant/lib/libimagequant.0.0.4.dylib'
  Referenced from: '/usr/local/Cellar/qmk/1.1.2_3/libexec/lib/python3.11/site-packages/PIL/_imaging.cpython-311-darwin.so'
  Reason: tried: '/usr/local/opt/libimagequant/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/local/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/local/Cellar/libimagequant/4.3.0/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/local/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/lib/libimagequant.0.0.4.dylib' (no such file)
```

### Keyboard Used

.

### Operating System

macOS 12.7.2

### qmk doctor Output

```text
james@James's MPB ~ % qmk doctor
Warning: Could not import qmk.cli.painter: ImportError, dlopen(/usr/local/Cellar/qmk/1.1.2_3/libexec/lib/python3.11/site-packages/PIL/_imaging.cpython-311-darwin.so, 0x0002): Library not loaded: '/usr/local/opt/libimagequant/lib/libimagequant.0.0.4.dylib'
  Referenced from: '/usr/local/Cellar/qmk/1.1.2_3/libexec/lib/python3.11/site-packages/PIL/_imaging.cpython-311-darwin.so'
  Reason: tried: '/usr/local/opt/libimagequant/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/local/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/local/Cellar/libimagequant/4.3.0/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/local/lib/libimagequant.0.0.4.dylib' (no such file), '/usr/lib/libimagequant.0.0.4.dylib' (no such file)
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /Users/james/qmk_firmware
Ψ Detected macOS 12.7.2 (Intel).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.23.9
Ψ - Latest master: 2024-02-16 18:35:10 -0500 (496d093fc3) -- [Keyboard] Add chlx ppr merro (#23014)
Ψ - Latest upstream/master: 2024-02-16 18:35:10 -0500 (496d093fc3) -- [Keyboard] Add chlx ppr merro (#23014)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-02-16 18:35:10 -0500 (496d093fc3) -- [Keyboard] Add chlx ppr merro (#23014)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

Making a copy of the existing libimagequant.0.4.0.dylib file and naming it libimagequant.0.0.4.dylib solves the crashing issue

fauxpark, 2024-02-19:
```
$ otool -L _imaging.cpython-311-darwin.so     
_imaging.cpython-311-darwin.so:
	/opt/homebrew/opt/jpeg-turbo/lib/libjpeg.8.dylib (compatibility version 8.0.0, current version 8.3.2)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.12)
	/opt/homebrew/opt/libtiff/lib/libtiff.6.dylib (compatibility version 7.0.0, current version 7.2.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1336.0.0)
```
libimagequant seems to not be referenced at all for me, though I'm on an M1 machine.

Please make sure brew is completely up to date, and run `brew reinstall qmk`.

lanceuye, 2024-02-22:
I get exact same error. I can provide other info if needed

fauxpark, 2024-02-23:
It looks like the Pillow dependency in the `qmk` formula needs updating. I figured out how to download the bottle directly, and yep, `otool -L` on its copy of the .so file above shows the wrong libimagequant reference.

Since last I checked there are no GHA runners for M1 (yet), so the CLI (more importantly Pillow) has to be compiled locally, which explains why I do not see it on my copy.

fauxpark, 2024-02-23:
Formula has been fixed, please update and try again.

JimmyBlackley, 2024-02-23:
Confirmed fixed, thanks

closed 2024-02-23 by fauxpark