qmk/qmk_firmware #23113: g_led_config mapping is missing keebio/iris/rev8/rev8.c

barnacker, 2024-02-19:
### Issue Description

The mapping is not working on the new board..
I tried the lazy way of using this one from rev7 on rev8 but its still wrong

Rev7:
https://github.com/qmk/qmk_firmware/blob/master/keyboards/keebio/iris/rev7/rev7.c#L17
[image]
[image]

My old code just lights the wrong leds now...
@nooges any ideas on how is the new board map should be now ?

barnacker, 2024-02-19:
ok that's helpful, I'm guessing this can be mapped out to make the old index numbers point to the new hardware layout... I'll investigate and see how this json works, thanks!

closed 2024-02-20 by barnacker

YosefBayoude, 2024-07-28:
It is true, the RGB configuration is found in the .json file now, but if for whatever reason you want the c version, here it is :
```c
led_config_t g_led_config = {{
    // Key Matrix to LED Index
    // Left Half
    { 0, 2, 3, 5, 6, 8},
    { 14, 13, 12, 11, 10, 9 },
    { 15, 16, 17, 18, 19, 20 },
    { 28, 26, 25, 23, 22, 21 },
    { NO_LED, NO_LED, 29, 30, 32, 33 },
    // Right Half
    { 34, 36, 37, 39, 40, 42 },
    { 48, 47, 46, 45, 44, 43 },
    { 49, 50, 51, 52, 53, 54 },
    { 62, 60, 59, 57, 56, 55 },
    { NO_LED, NO_LED, 63, 64, 66, 67 }
}, {
    // LED Index to Physical Position
    // Left Half
    {0, 5}, {8, 5}, {16, 5}, {32, 2}, {40, 1}, {48, 0}, {64, 5}, {72, 3}, {80, 3},
    {80, 17}, {64, 15}, {48, 13}, {32, 15}, {16, 18}, {0, 18},
    {0, 32}, {16, 32}, {32, 28}, {48, 27}, {64, 28}, {80, 30},
    {80, 43}, {64, 42}, {48, 40}, {40, 50}, {32, 42}, {16, 45}, {8, 43}, {0, 45},
    {98, 52}, {90, 64}, {80, 58}, {72, 58}, {56, 47},

    // Right Half
    {224, 5}, {216, 5}, {208, 5}, {192, 2}, {184, 1}, {176, 0}, {160, 5}, {152, 3}, {144, 3},
    {144, 17}, {160, 15}, {176, 13}, {192, 15}, {208, 18}, {224, 18},
    {224, 32}, {208, 32}, {192, 28}, {176, 27}, {160, 28}, {144, 30},
    {144, 43}, {160, 42}, {176, 40}, {184, 50}, {192, 42}, {208, 45}, {216, 43}, {224, 45},
    {168, 47}, {152, 58}, {144, 58}, {134, 64}, {126, 52}
}, {
    // LED Index to Flag
    // Left Half
    4, 2, 4, 4, 2, 4, 4, 2, 4,
    4, 4, 4, 4, 4, 4,
    4, 4, 4, 4, 4, 4,
    4, 4, 4, 2, 4, 4, 2, 4,
    4, 4, 2, 4, 4,

    // Right Half
    4, 2, 4, 4, 2, 4, 4, 2, 4,
    4, 4, 4, 4, 4, 4,
    4, 4, 4, 4, 4, 4,
    4, 4, 4, 2, 4, 4, 2, 4,
    4, 4, 2, 4, 4

} };

```