qmk/qmk_firmware #23132: make failes for some values of processor in info.json

zegonix, 2024-02-22:
### Describe the Bug

For a lack of qmk configuration for the tkd cycle 7 I tried doing it myself. When choosing cortex-m3 for the processor key in my info.json (chips is apm32f103), the build fails. It also fails when choosing stm32f103, which might be compatible, but I was to lazy to check. I also tried stm32f401 because I recently built a firmware for my mode envoy with that chip and interestingly enough it builds without errors.

here is the command output:
>> qmk compile
Ψ Compiling keymap with make --jobs=4 --output-sync=target -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=zegonix/cycle7 KEYMAP=default KEYBOARD_FILESAFE=zegonix_cycle7 TARGET=zegonix_cycle7_default INTERMEDIATE_OUTPUT=.build/obj_zegonix_cycle7_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

platforms/chibios/platform.mk:102: lib/chibios-contrib/os/common/startup/ARMCMx/compilers/GCC/mk/startup_.mk: No such file or directory
make: *** No rule to make target 'lib/chibios-contrib/os/common/startup/ARMCMx/compilers/GCC/mk/startup_.mk'.  Stop.
make: *** Waiting for unfinished jobs....
Generating: .build/obj_zegonix_cycle7_default/src/info_deps.d                                       [OK]

### Keyboard Used

custom

### Link to product page (if applicable)

https://clickclack.io/pages/tkd-cycle7

### Operating System

manjaro (xfce)

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/zegonix/qmk_firmware
Ψ Detected Linux (Manjaro Linux).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.23.2
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2023-12-27 12:12:42 +1100 (cc08f230ff) -- bm40hsrgb/rev2: disable some RGB Matrix effects to reduce filesize (#22761)
Ψ - Latest upstream/master: 2024-02-21 14:35:39 -0800 (4e04da397e) -- [Keyboard] Add the Compensator (#23103)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-12-27 12:12:42 +1100 (cc08f230ff) -- bm40hsrgb/rev2: disable some RGB Matrix effects to reduce filesize (#22761)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.2.0
Ψ Found avr-gcc version 13.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 1.0.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

fauxpark, 2024-02-23:
`cortex-m3` on its own is insufficient, you need to set a few other things. This is only used when the MCU being used does not match anything else. In this case other boards already in the repo which use APM32F103 simply set `STM32F103` as it is a drop-in replacement.

zegonix, 2024-02-23:
well that makes sense, but as I descibed, STM32F103 doesn't comile either.. it expects a folder <chibios-contrib>/os/hal/boards/GENERIC_STM32_F103 which does not exist.

sigprof, 2024-02-23:
STM32F103 requires some kind of a user bootloader to support firmware updates over USB, therefore a bare STM32F103 target would not be very useful (you would need to use ST-Link or maybe some serial port adapter to flash the firmware in that case).  And there are multiple bootloader implementations for STM32F103, which differ in size and therefore require different LD files to build the firmware; one of those implementation must be chosen explicitly.

Yes, this makes STM32F103 an outlier; all other supported MCUs fall in one of the following categories:
- Some chips have builtin bootloaders with USB support (most supported STM32 models, RP2040).
- Most AVR chips have separate flash sections for the bootloader and the main firmware, which makes the main firmware mostly independent from the bootloader implementation, therefore compiling the firmware without specifying a bootloader explicitly usually gives something useful (although some features, like `QK_BOOT`, may fail if the actual bootloader differs from the assumed one).
- For some other AVR chips like ATtiny85, which don't have a dedicated flash section for the bootloader, only a single bootloader implementation is supported.
- Although Kinetis MCUs used on 32-bit Teensy boards don't have a builtin bootloader, those Teensy boards have a second small MCU which can upload the bootloader to the main MCU over the debug interface when the flash button is pressed, therefore having a configuration for a bare MCU is useful for those boards.

zegonix, 2024-02-23:
alright, thanks for the information. i'll read into it tomorrow.