qmk/qmk_firmware #23134: how to flash pro micro via arduino uno as isp?

idk-br9, 2024-02-23:
### Issue Description

I have pro micro without usb connector, and i need to flash it. I also have arduino uno with arduinoISP sketch.
When i trying to flash controller qmk says: "programmer is not responding". How i can fix it?

fauxpark, 2024-02-23:
Please read the [ISP flashing guide](https://docs.qmk.fm/#/isp_flashing_guide)

idk-br9, 2024-02-23:
> Please read the [ISP flashing guide](https://docs.qmk.fm/#/isp_flashing_guide)

I initially did everything according to this guide

idk-br9, 2024-02-23:
> Post some pictures of your hardware and wiring, or join the [QMK discord](https://discord.gg/JmczBgfj) and share images there

I cant give photo, but my wiring is the same as wiring in tutorial

closed 2024-05-26 by tzarc