qmk/qmk_firmware #23145: Trying to use my keyboard with SignalRGB

LywenNagaline, 2024-02-24:
### Issue Description

Hi guys, 

I recently bought the K5 Skillkorp keyboard (from Boulanger brand and Asus Rog manufacturer). 
When I use SignalRGB to manage the RGB of the keyboard, it is recognized as: GMMK 2 96% ANSI QMK Keyboard. 
In fact it is detected but not controllable and I couldn't find any compatible QMK firmware to use it with SignalRGB. 

Could you help me ? 

Thanks ,

Lywen
 
[image]

zvecr, 2024-02-24:
Support is maintained in another fork and is not part of QMK yet.
You may need to join their #roles channel and add QMK to your list.

Site: https://signalrgb.com/
Git: https://github.com/SRGBmods/QMK
Discord: https://discord.com/invite/J5dwtcNhqC
Docs: https://docs.signalrgb.com/qmk/qmk-introduction

closed 2024-02-24 by zvecr

LywenNagaline, 2024-02-24:
Actually, I've already chatted with them on their discord on the QMK channels, and they're the ones who told me to come straight here and ask you.
They told me to find the QMK source to add it in their software.
So here I am.

keyboard-magpie, 2024-02-24:
We have no source code for a keyboard called a "K5 Skillkorp". Given that it is appearing as another keyboard, the best we can say is that you have some dreadful manufacturer spoofing the information a GMMK board would send. 

We cannot provide source code arbitrarily; it is down to manufacturers to produce this or kind-spirited community members to reverse engineer boards.

You bought a no-name board with no good firmware support, sorry.

LywenNagaline, 2024-02-24:
Thank you for your answer. 
At least now I know that I don't have to looking for an inexistant firmware. 

Have a nice day

zvecr, 2024-02-24:
> Why do you close the topic ?

Because it is off topic for this repository.

> . If you are on their discord you should see the whole conversation.

I'm not going to go chase down a discussion on something we do not support.

> You didn't even helped me

We are not obligated to provide help for every random fork of QMK that we have zero control over. That is why you were redirected to the correct location.

> How rude :(

Maybe I could have written more words than the Discord bot is configured with. However that takes time away from performing actual maintenance on QMK. The QMK firmware repository averages 300 PRs opened and 300 PRs merged every month. This particular point in time is right at the end of the current [develop](https://docs.qmk.fm/#/breaking_changes?id=when-is-the-next-breaking-change) cycle. Had your question been "How do I get QMK source for my keyboard", it would have still been closed with a similar message. 

> Please refrain from asking maintainers to add support for specific keyboards -- it is unlikely they will have hardware available, and will not be able to help.
Your best bet is to take the initiative, add support, then submit a PR yourself.