qmk/qmk_firmware #23191: [Bug] Encoder on slave continues to trigger forever once activated

jasonish, 2024-03-01:
### Describe the Bug

After updating from an older version to latest git and flashing both halves of my keyboard, the non-primary side encoder, once rotated continues to trigger forever at a rate that causes the keyboard to become unresponsive.

The primary side operates as you might expect. 

I've narrowed it down to commit 9d9cdaaa2d035787b0b50c26f2975695fdbc16f4.
The commit before, 2eb9ff8efd1df2c98724481c71c8ab8a5b62e31e functions OK.

### Keyboard Used

sofle_choc

### Link to product page (if applicable)

Linux

### qmk doctor Output

```text
QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/jason/keyboards/qmk/master
Ψ Detected Linux (Fedora Linux 39 (Workstation Edition)).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.24.0
Ψ - Latest master: 2024-02-28 21:47:37 +1100 (4e369d405af) -- Merge branch 'develop'
Ψ - Latest upstream/master: 2024-02-28 21:47:37 +1100 (4e369d405af) -- Merge branch 'develop'
Ψ - Latest upstream/develop: 2024-02-28 19:13:04 -0700 (4e953f11696) -- Fixup mechlovin/octagon (#23179)
Ψ - Common ancestor with upstream/master: 2024-02-28 21:47:37 +1100 (4e369d405af) -- Merge branch 'develop'
Ψ - Common ancestor with upstream/develop: 2024-02-28 21:47:37 +1100 (4e369d405af) -- Merge branch 'develop'
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.1.0
Ψ Found avr-gcc version 13.1.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f90)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

yottta, 2024-03-03:
➕ Same issue here.
Keyboard: Sofle
OS: Mac/Win
Issue visible on the primary instead.

haunt98, 2024-03-04:
Same issue,
Keyboard: Sofle V2 with converter rp2040
OS: Mac/Win
Both primary, slave rotated got trigger

Coko7, 2024-03-05:
Same issue as well.
Keyboard: `sofle/rev1`
OS: `Linux/Win`
Primary encoder (left) is fine, but secondary (right) gets triggered repeatedly, as described by OP.

tzarc, 2024-03-05:
@jasonish @yottta @Coko7 are you using RP2040 converters? Or just normal 32U4-based?

Coko7, 2024-03-05:
@tzarc I believe I am using an ATmega32U4, although I need to confirm it.

jasonish, 2024-03-05:
> @jasonish @yottta @Coko7 are you using RP2040 converters? Or just normal 32U4-based?

ATmega32U4

tzarc, 2024-03-06:
Have updated #23195 with some potential ways forward, would be good to get some confirmation as to whether it's headed in the right direction. Still can't reproduce locally, so basically this is just some defensive coding.

Coko7, 2024-03-07:
@tzarc Thank you for the work!
I have tried it out just now, and it seems to fix it on my keyboard. I have set up my secondary encoder for page up/down motion, and now it behaves as it should (without infinitely triggering).
I will test it more today and come back here if I run into some issues.

yottta, 2024-03-07:
> @tzarc installed it. Will keep you posted. Thanks a lot for the great effort! 👏

After 1 day I see no other issues. Thanks a lot for it.

PR #23195 (2024-03-10): Fixes for encoder abstraction.
## Description

As reported on Discord ([here onwards](https://discord.com/channels/440868230475677696/440868230475677698/1212999470619566100)), and subsequently #23191.

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* (Hopefully) Fixes #23191.

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Pytho [... cut]
closed 2024-03-10 by tzarc