qmk/qmk_firmware #23197: [Bug] via keymap compile error on macos (Apple M2 Pro)

hitsmaxft, 2024-03-02:
### Describe the Bug

```
qmk compile -kb 'keychron/q60/ansi' -km via
```
output 

```
❯ qmk compile -kb 'annepro2/c18' -km via
Ψ Compiling keymap with make -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=annepro2/c18 KEYMAP=via KEYBOARD_FILESAFE=annepro2_c18 TARGET=annepro2_c18_via INTERMEDIATE_OUTPUT=.build/obj_annepro2_c18_via VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

arm-none-eabi-gcc (Arm GNU Toolchain 12.3.Rel1 (Build arm-12.35)) 12.3.1 20230626
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: quantum/via.c                                                                            [OK]
Linking: .build/annepro2_c18_via.elf                                                                [ERRORS]
 |
 | /nix/store/f4ci6pixmp1irjayfw8v0pgqcjf279f8-gcc-arm-embedded-12.3.rel1/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld:rules_memory.ld:314 cannot move location counter backwards (from 200020a8 to 20001ffc)
 | collect2: error: ld returned 1 exit status
 |
make: *** [builddefs/common_rules.mk:280: .build/annepro2_c18_via.elf] Error 1
```

uname -a 
```
Darwin xxxxxxxxxxxxxxxx 23.0.0 Darwin Kernel Version 23.0.0: Fri Sep 15 14:43:05 PDT 2023; root:xnu-10002.1.13~1/RELEASE_ARM64_T6020 arm64 arm Darwin
```

### Operating System

Darwin qx-workstation.local 23.0.0 Darwin Kernel Version 23.0.0: Fri Sep 15 14:43:05 PDT 2023; root:xnu-10002.1.13~1/RELEASE_ARM64_T6020 arm64 arm Darwin

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /Users/qixiang/Projects/qmk_firmware
Ψ Detected macOS 14.0 (Apple Silicon).
Ψ Userspace enabled: False
Ψ Git branch: feature/annepro-via
Ψ Repo version: 0.24.0
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.3.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb16)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f9)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2024-03-02:
It won't be _any_ via keymap, just the annepro2.
Your compiler is producing binaries too big for the MCU.

Turn off some RGB effects or something.

hitsmaxft, 2024-03-02:
> It won't be _any_ via keymap, just the annepro2.
> 
> Your compiler is producing binaries too big for the MCU.
> 
> 
> 
> Turn off some RGB effects or something.

thank you， i will check it later.
but it did reproducing on compiling keychon  in my macOS .

hitsmaxft, 2024-03-03:
> It won't be _any_ via keymap, just the annepro2. Your compiler is producing binaries too big for the MCU.
> 
> Turn off some RGB effects or something.

you're right ,  reducing functions works

closed 2024-03-03 by hitsmaxft

turkishfilms, 2024-03-19:
i am having similar troubles. i have a piantor and tapdance works just fine when i use vial, but im trying to code it myself in qmk. please help, my fork has it under turkish

hitsmaxft, 2024-03-20:
> i am having similar troubles. i have a piantor and tapdance works just fine when i use vial, but im trying to code it myself in qmk. please help, my fork has it under turkish

you need to start a new issue and post error message like this issue. 

if error message is same as this issue. you may disable features as much as possible, then enable one by one to test, rgb matrix, mouse key etc