qmk/qmk_firmware #23207: [Bug] Error occurs while trying to compile production target keyboard firmware

MaxJ345, 2024-03-04:
### Describe the Bug

I'm trying to build production target keyboard firmware using the QMK CLI. When I do so, I receive an error during the compilation process:

```
$ qmk compile --keyboard project_vanguard/full_size/v1 --keymap default --target production

...

c:/qmk_msys/mingw64/bin/../lib/gcc/avr/12.2.0/../../../../avr/bin/ld.exe: section .apitable_trampolines LMA [00007fa0,00007fb7] overlaps section .data LMA [00007f2a,00007fa1]
collect2.exe: error: ld returned 1 exit status
make[1]: *** [../../LUFA/Build/DMBS/DMBS/gcc.mk:251: BootloaderDFU.elf] Error 1
make: *** [platforms/avr/platform.mk:214: bootloader] Error 2
```

This isn't just a problem with my keyboard. I've also noticed it with other keyboards as well. For example, the following two will produce a similar error:

```
$ qmk compile --keyboard evyd13/gh80_1800 --keymap default --target production

...

c:/qmk_msys/mingw64/bin/../lib/gcc/avr/12.2.0/../../../../avr/bin/ld.exe: section .apitable_trampolines LMA [00007fa0,00007fb7] overlaps section .data LMA [00007f36,00007fa9]
collect2.exe: error: ld returned 1 exit status
make[1]: *** [../../LUFA/Build/DMBS/DMBS/gcc.mk:251: BootloaderDFU.elf] Error 1
make: *** [platforms/avr/platform.mk:214: bootloader] Error 2
```

```
$ qmk compile --keyboard evyd13/wasdat_code --keymap default --target production

...

c:/qmk_msys/mingw64/bin/../lib/gcc/avr/12.2.0/../../../../avr/bin/ld.exe: section .apitable_trampolines LMA [00007fa0,00007fb7] overlaps section .data LMA [00007f2a,00007fa1]
collect2.exe: error: ld returned 1 exit status
make[1]: *** [../../LUFA/Build/DMBS/DMBS/gcc.mk:251: BootloaderDFU.elf] Error 1
make: *** [platforms/avr/platform.mk:214: bootloader] Error 2
```

Here is the entire log from an example build:

```
$ qmk clean --all
QMK Firmware 0.24.1
Deleting .build/ ... done.
Deleting *.bin, *.hex, and *.uf2 ... done.

$ qmk compile --keyboard project_vanguard/full_size/v1 --keymap default --target production
Ψ Compiling keymap with make -r -R -f builddefs/build_keyboard.mk -s production KEYBOARD=project_vanguard/full_size/v1 KEYMAP=default KEYBOARD_FILESAFE=project_vanguard_full_size_v1 TARGET=project_vanguard_full_size_v1_default INTERMEDIATE_OUTPUT=.build/obj_project_vanguard_full_size_v1_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

Generating: .build/obj_project_vanguard_full_size_v1_default/src/info_deps.d                        [OK]
avr-gcc.exe (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_project_vanguard_full_size_v1_default/src/info_config.h                      [OK]
Generating: .build/obj_project_vanguard_full_size_v1_default/src/default_keyboard.c                 [OK]
Generating: .build/obj_project_vanguard_full_size_v1_default/src/default_keyboard.h                 [OK]
Compiling: keyboards/project_vanguard/full_size/v1/v1.c                                             [OK]
Compiling: .build/obj_project_vanguard_full_size_v1_default/src/default_keyboard.c                  [OK]
Compiling: quantum/keymap_introspection.c                                                           [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                       [OK]
[... 137 lines cut ...]
Ψ - Common ancestor with upstream/develop: 2024-03-02 08:45:01 +1100 (1875659df0) -- CLI Speed improvements. (#23189)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f90)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

zvecr, 2024-03-04:
Issue is due to differences in avr-gcc compiler version.

```
qmk compile --keyboard evyd13/gh80_1800 --keymap default --target production
Ψ Compiling keymap with make -r -R -f builddefs/build_keyboard.mk -s production KEYBOARD=evyd13/gh80_1800 KEYMAP=default KEYBOARD_FILESAFE=evyd13_gh80_1800 TARGET=evyd13_gh80_1800_default INTERMEDIATE_OUTPUT=.build/obj_evyd13_gh80_1800_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk" QMK_USERSPACE=/home/zvecr/qmk_userspace

Generating: .build/obj_evyd13_gh80_1800_default/src/info_deps.d                                     [OK]
avr-gcc (GCC) 13.2.0
...
AVR Memory Usage
----------------
Device: atmega32u4

Program:    3964 bytes (12.1% Full)
(.text + .data + .bootloader)

Data:        155 bytes (6.1% Full)
(.data + .bss + .noinit)

BootloaderDFU.hex copied to evyd13_gh80_1800_default_bootloader.hex
File sizes:
   text	   data	    bss	    dec	    hex	filename
      0	  16024	      0	  16024	   3e98	evyd13_gh80_1800_default.hex
      0	   4020	      0	   4020	    fb4	evyd13_gh80_1800_default_bootloader.hex
      0	  20044	      0	  20044	   4e4c	evyd13_gh80_1800_default_production.hex
```

8.x has been mentioned as good, 13 as used above builds fine.

MaxJ345, 2024-03-04:
Should I update avr-gcc using pacman?: `pacman -S avr-gcc`

Or is there some other/preferred way of doing it using the QMK CLI?

MaxJ345, 2024-03-04:
Actually, I don't see avr-gcc listed by pacman as an installed package.

How do I set the version of avr-gcc used by QMK?

zvecr, 2024-03-04:
<https://packages.msys2.org/package/mingw-w64-x86_64-avr-gcc> `12.2.0` is the latest available for MSYS.

MaxJ345, 2024-03-04:
So how did you get QMK to compile with avr-gcc version 13.2.0 above?

tzarc, 2024-03-06:
Put simply, I don't use QMK MSYS; different Linux distributions bundle different compiler versions, and in my case, I've compiled my own copy of gcc 13.2.0 to run under Linux.

Ultimately this is outside the scope of QMK's support -- we make a best-effort attempt at getting things to work on as many operating system combinations as possible, but with the number of people we have and the numerous variations in the wild, we can't support everything. We don't have control over which version of GCC is packaged by the MSYS team, and at the moment I'd hazard a guess and say that almost nobody can build production targets under QMK MSYS.

If it's critical that you build production targets, your best bet is to figure out an alternative compilation environment such as Linux. If you can run Docker locally you can try to use the prebuilt QMK CLI container which contains an older version of GCC (in the 8.x series) which under normal circumstances should work fine. Neither of these solutions are going to be as simple to get working as QMK MSYS.

MaxJ345, 2024-03-07:
I see. That's unfortunate.

I'll try to see if I can get it working in a Linux-based environment. Thank you for your help!

MaxJ345, 2024-03-11:
I setup a Linux-based environment (specifically Manjaro) and was finally able to compile production firmware. The versions of avr-gcc used were 8.3.0 and 13.2.0.

Thanks for the help!

closed 2024-03-11 by MaxJ345

dotleon, 2024-07-03:
Downgrading avr-gcc solved this for me.

At the time of writing this, avr-gcc 8.5.0 for MSYS2 is available at [msys2 repo site](https://repo.msys2.org/mingw/mingw64/) (mingw-w64-x86_64-avr-gcc-8.5.0-1-any.pkg.tar.zst)

Then you can downgrade with
`pacman -U mingw-w64-x86_64-avr-gcc-8.5.0-1-any.pkg.tar.zst`

After this, production compiles.