qmk/qmk_firmware #23209: [Feature Request] Define the matrix scan interval to adjust key press delay

boulevard-h, 2024-03-04:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hello,

I am working with the QMK firmware and I'm looking to customize the behavior of my keyboard. Specifically, I want to adjust the key press delay by defining the interval at which the keyboard matrix is scanned. I've searched through the QMK docs and codebase, but I have not found any macros or functions that seem to allow this adjustment.

Could you please provide guidance on how to define the matrix scan interval, or point me to the relevant part of the documentation if it exists?

Thank you for your assistance.

myst729, 2024-03-12:
What are you going to do?

If you want to [change the default tap-hold threshold](https://docs.qmk.fm/#/tap_hold?id=tapping-term), try `#define TAPPING_TERM`. If your want to [deal with the physical switch contact bounce](https://docs.qmk.fm/#/feature_debounce_type?id=debounce-time), try `#define DEBOUNCE`.

As far as I know, you can [measure the matrix scan frequency](https://docs.qmk.fm/#/faq_debug?id=how-long-did-it-take-to-scan-for-a-keypress), but you cannot explicitly set the scan interval. I guess part of the reason is [QMK supports wide range of different MCUs](https://docs.qmk.fm/#/compatible_microcontrollers?id=compatible-microcontrollers) which vary significantly in performance. However, there's a [hardware option](https://docs.qmk.fm/#/config_options?id=hardware-options) to modify the delay between changing matrix pin state and reading values, try `#define MATRIX_IO_DELAY`. After all, a typical matrix scan is just a series of "changing row pin state and reading column pin values"

drashna, 2024-05-28:
There is also the dynamic tapping term feature too:
https://docs.qmk.fm/#/tap_hold?id=dynamic-tapping-term

closed 2024-09-26 by github-actions[bot]