qmk/qmk_firmware #23226: [Bug]  AnnePro2 'g' key scan issue

hitsmaxft, 2024-03-06:
### Describe the Bug

compiling annepro2 fw from master would makes key 'G' very hard to press.
after merging branch annepro2_core_matrix , key 'G' is easier to press , but some times missing

while long press G didn't make it input , so it's not a tap or delay input. 
press very very hard always make it input 

this problem will gone while flash stock firmware, so its a  fw issue, not hardware bug.

### Keyboard Used

annepro2/c18

### Operating System

macos

### qmk doctor Output

```text
❯ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /Users/qixiang/Projects/keyboards/qmk_firmware
Ψ Detected macOS 14.3.1 (Apple Silicon).
Ψ Userspace enabled: False
Ψ Git branch: annepro2_core_matrix
Ψ Repo version: 0.24.2
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.3.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb16)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f9)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ - keyboards/annepro2/keymaps/macvim: 2024-03-05 20:03:29 +0800 --  (a9c190c)
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

no

hitsmaxft, 2024-03-06:
@zvecr  thanks for you pr make annepro2 better, is there any chance to make it better.

zvecr, 2024-03-06:
> is there any chance to make it work perfectly again

Given I cannot reproduce the issue, and I dont use the keyboard outside testing purposes, probably not.

Some areas to look into:

* Different compiler version?
* Larger values of `MATRIX_IO_DELAY` over the default of `30`?
* Any Intermittent communication issues with uart
  * Disable/rip out bluetooth and rgb_matrix code?

My primary concern of #23221 was to remove completely redundant code (that I had previously said was not required at the time of the keyboards PR). Its unlikely I would look into the issue myself.

tzarc, 2024-03-06:
Just confirming that I can't reproduce the issue with the 'g' key on my AP2/C18 either.

hitsmaxft, 2024-03-07:
> > is there any chance to make it work perfectly again有没有机会让它再次完美运行
> 
> Given I cannot reproduce the issue, and I dont use the keyboard outside testing purposes, probably not.鉴于我无法重现该问题，并且我在测试目的之外不使用键盘，可能不会。
> 
> Some areas to look into:一些需要研究的领域：
> 
>     * Different compiler version?不同的编译器版本？
> 
>     * Larger values of `MATRIX_IO_DELAY` over the default of `30`? `MATRIX_IO_DELAY` 的值大于默认值 `30` 吗？
> 
>     * Any Intermittent communication issues with uart
>       与 uart 的任何间歇性通信问题
>       * Disable/rip out bluetooth and rgb_matrix code?禁用/删除蓝牙和 rgb_matrix 代码？
> 
> 
> My primary concern of #23221 was to remove completely redundant code (that I had previously said was not required at the time of the keyboards PR). Its unlikely I would look into the issue myself.我对 #23221 的主要关注是删除完全冗余的代码（我之前说过在键盘 PR 时不需要）。我不太可能亲自调查这个问题。

Since there are other user in annepro2 discord reproduce this issue too, maybe a version related issue of annepro2 itself
I will try your suggestions . Thanks again.

hitsmaxft, 2024-03-07:
update: installing same firmware on my another annepro2 c18, nothing wrong happened, so it's a hardware related but likely fix by stock firmware

lukevanlukevan, 2024-03-11:
Joining to say I experience the same issue, `g` and `1` key are intermittent but stock firmware does work. Maybe a bad switch that now operates under a certain threshold when using QMK that the main firmware still reads correctly.

hitsmaxft, 2024-03-11:
> Joining to say I experience the same issue, `g` and `1` key are intermittent but stock firmware does work. Maybe a bad switch that now operates under a certain threshold when using QMK that the main firmware still reads correctly. 

I've fix the issue by config debounce to 3ms on developer (which just updated annepro2' matrix scan code)

this problem did triggered by some hardware  related issue. but lower debounce works on my keyboard. you may try it

hitsmaxft, 2024-03-12:
> Given I cannot reproduce the issue, and I dont use the keyboard outside testing purposes, probably not.鉴于我无法重现该问题，并且我在测试目的之外不使用键盘，可能不会。
>
> Some areas to look into:一些需要研究的领域：
>
>     * Different compiler [...]

I just figure it out that lower debounce will fix this problem , but since it not a common issue , may i create a pull request , just add solution into  read me instead of  apply it directly to the config.h

hitsmaxft, 2024-03-12:
solution found

closed 2024-03-12 by hitsmaxft

Fede-26, 2026-11-11:
This may be worth another look — I think there is a mechanism here that fits, though I have not reproduced this specific issue.

The Anne Pro 2 matrix scans at exactly **200 Hz**, and not because of the matrix. `matrix_output_unselect_delay()` calls `wait_us(MATRIX_IO_DELAY)` once per row, and on this board `wait_us` is not a busy wait: `WAIT_US_TIMER` is undefined, and `PORT_SUPPORTS_RT` is false because the HT32F523xx is a Cortex-M0 and ARMv6-M has no DWT cycle counter. So it falls through to `chThdSleepMicroseconds(30)`, which with `CH_CFG_ST_FREQUENCY` at 1000 rounds up to a whole 1 ms tick. Five rows, 5 ms, 200 Hz.

Debouncing works on sampled state, so at 200 Hz the default `DEBOUNCE` of 5 ms is a **single confirming sample**. With `sym_defer_g` the whole matrix additionally has to be change-free for that window. A contact that is intermittent rather than cleanly closed — which is what "it works if I press very very hard" describes — can plausibly fail to settle within one sample and get dropped, at some rate rather than always. That would also fit "nothing wrong on my other C18": it needs a marginal switch to show up, which is why it looks hardware-dependent while still being a firmware problem.

#26573 gives `wait_us` a hardware timer; measured 200 Hz → 3304 Hz on a C18. On my board the chattering this caused was severe before and gone after.

Again: this is a hypothesis for *this* issue, not a diagnosis — I do not have the affected hardware. But it is cheap to retest on top of that change.