qmk/qmk_firmware #23255: keyboard_protocol stuck to 0 after OS reboot/start on certain setups

FabienFellay, 2024-03-10:
### Describe the Bug

This issue was actually tracked by issue #2248 but was later set as completed because of a lack of activity. However, the issue is real. On some computers (like mine), when the computer boots with the keyboard plugged in, then clearly, `keyboard_protocol` get stuck to 0, which leads to a non-working NKRO. The only way to recover is to unplug and plug again the keyboard (or to issue a QK_REBOOT if such a key is present in the `keymap.c`). Then, `keyboard_protocol` finally gets to 1. At that moment, if `keymap_config.nkro` is set to 1, we get a working NKRO.

I could work-around the bug by forcing keyboard_protocol to 1 at line 677 of `tmk_core/protocol/chibios/usb_main.c`:

                            keyboard_protocol = setup->wValue.word;

was replaced by:

                            keyboard_protocol = 1;

https://github.com/qmk/qmk_firmware/blob/3e1ac7a38fa4e6885053a762bc75f7c4e068eccb/tmk_core/protocol/chibios/usb_main.c#L677C3-L677C68

In that case, `keyboard_protocol` is 1 right after a computer boot and NKRO is immediately working. This only confirms the bug but this could not be considered as a valid solution. Maybe this bug is quite widely spread but has been unnoticed because must people do not really care about NKRO (or `keyboard protocol`) if all other features of the keyboard are working.

I also tried the following without success:

* Add `#define FORCE_NKRO` in `config.h`: as expected, this forces `keymap_config.nkro` to 1 when the keyboard boot but this has no impact on the rootcause `keyboard protocol` that stays to 0.
* Add `WAIT_FOR_USB = yes` in `rule.mk`
* Add `#define USB_SUSPEND_WAKEUP_DELAY 1000` in `config.h` (much larger values has been tried as well with no success)
* Add `NO_USB_STARTUP_CHECK = yes` in `rule.mk`
* Add `MOUSE_SHARED_EP = no` and `KEYBOARD_SHARED_EP = no` in `rule.mk`

Non of the above list items worked, only forcing `keyboard_protocol` to 1 in `tmk_core/protocol/chibios/usb_main.c`.

If no solution is found, **qmk** could propose a new `#define` to the user in order to cleanly force `keyboard_protocol` to 1. However, the side effect will probably be a non-working keyboard in some BIOS (which is not ideal).

Another proposal could be: as long as `keyboard_protocol` has been determined to be 0, the keyboard should regularly try (i.e. every 5 sec or so) to assess if it is OK to switch it to 1.

I would love to have this fixed, this bug has been there for years.

### Keyboard Used

Keychron q6

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-q6-qmk-custom-mechanical-keyboard

### Operating System

Windows 10

### Is AutoHotKey / Karabiner installed

- [X] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

FabienFellay, 2024-04-07:
I have now more hints about how to reproduce the issue.

I have 2 identical machines: one with Windows 10 only and one with a dual boot Windows 10 / Ubuntu. The above-described bug only affects the machine that has the dual boot. I don't know what causes the issue in the first place, but the dual boot machine may shortly pass through a transient (not really visible) bios-like menu that somehow makes the keyboard being stuck with `keyboard_protocol` to 0.

Can a QMK collaborator reproduce the issue based on those new elements (machines with dual-boot) ?

danei, 2024-04-27:
@FabienFellay For grub, it sometimes requires toggling legacy USB support in BIOS/UEFI.
Grub itself checks boot class and protocol from descriptor when initializing usb keyboard, and aborts when mismatch:
```
if (usbdev->descdev.class != 0 
      || usbdev->descdev.subclass != 0 || usbdev->descdev.protocol != 0)
    return 0;

  if (usbdev->config[configno].interf[interfno].descif->subclass
      != USB_HID_BOOT_SUBCLASS
      || usbdev->config[configno].interf[interfno].descif->protocol
      != USB_HID_KBD_PROTOCOL)
    return 0;
```
The logic very likely differs from BIOS/UEFI. If grub hands over as is to OS, it's possible OS won't re-initialize usb keyboard again as grub has initialized it.

FabienFellay, 2024-04-30:
@danei Thank you very much! OS detection + reset feature helped: adding `OS_DETECTION_ENABLE = yes` in rules.mk and `#define OS_DETECTION_KEYBOARD_RESET` in my config.h just solved this issue.

Indeed, all my dual-boot machines that were impacted by this issue are now showing a `keyboard_protocol` to 1 right after the OS boots and NKRO is finally working directly without an additional manual reset.

I let a link to this issue in the issue #2248 (this might help some people).

Thank you! :thumbsup:

closed 2024-04-30 by FabienFellay