qmk/qmk_firmware #23261: Clarification on accessing the bootloader

MaxJ345, 2024-03-12:
### Issue Description

Hello,

I would like some clarification on how to access the bootloader in QMK firmware.

For context, I built QMK production firmware with the QMK DFU bootloader:

```
qmk clean --all

qmk compile --keyboard project_vanguard/full_size/v1 --keymap default --target production
```

I set the fuses to:
* `lfuse` = `0xFF`
* `hfuse` = `0x98` and `0x99`
* `efuse` = `0xFB`

Here is my code for reference: https://github.com/MaxJ345/qmk_firmware/compare/master...MaxJ345:qmk_firmware:test

My results are:

* With the high fuse set to `0x98`:
    * Physical reset (shorting the `RESET` pin to ground) --> bootloader mode
    * `QK_BOOT` keycode --> keyboard application firmware

* With the high fuse set to `0x99`:
    * Physical reset (shorting the `RESET` pin to ground) --> keyboard application firmware
    * `QK_BOOT` keycode --> bootloader mode

Are things behaving correctly here? I thought both a physical reset and the `QK_BOOT` keycode would reset the microcontroller into bootloader mode. Am I misunderstanding how it's supposed to work?

sigprof, 2024-03-12:
The intended fuse settings for the `atmel-dfu` and `qmk-dfu` bootloaders are described in the [ISP flashing guide](https://docs.qmk.fm/#/isp_flashing_guide?id=atmel-dfu); however, that guide does not give a detailed explanation of those settings and the corresponding hardware configuration:
- `hfuse` = `0x99` (the `BOOTRST` fuse is not programmed), or `0xD9` if you also want to disable JTAG immediately (but the QMK startup code would disable JTAG during early startup anyway).
- `efuse` = **`0xF3`** (the `HWBE` fuse **is programmed**).
- The PE2/HWB pin is pulled **down** (preferably with a resistor, so that flashing a firmware which tries to use the `E2` pin as output won't overload the pin, but a direct connection to GND could work too — apparently the reference Pro Micro design have it connected like that).
- The RESET pin must not have a capacitor connected, so that RESET does not get pulled low when power is applied.

The original `atmel-dfu` bootloader cannot handle the configuration with `BOOTRST` enabled (the chip would be stuck in the bootloader).  There is some code in `qmk-dfu` which attempts to handle `BOOTRST`, but apparently that code does not work correctly.  Instead of using `BOOTRST`, those bootloaders rely on `HWBE` (and a somewhat underdocumented detail that the state of the HWB pin does not affect the behavior of the power-on reset).

The requirement to pull the `E2` pin down prevents the usage of that pin in the matrix, but you can still use that pin for some tasks that are compatible with the pull-down — e.g., lots of ATmega32U4-based boards use that pin as `WS2812_DI_PIN`.

MaxJ345, 2024-03-12:
That pin (`PE2/HWB`) is currently not connected to anything in my design. That's probably the source of my problem.

My `RESET` pin is connected to `VCC` via a pull-up resistor. Is that fine?

sigprof, 2024-03-12:
Yes, having the PE2/HWB pin floating won't work if you intend to enable the `HWBE` fuse; however, your `efuse` value also does not enable `HWBE`.

Having a pull-up on the RESET pin is fine; but some people also add a capacitor “just for a good measure” (and, e.g., STM32 datasheets recommend having a capacitor on the NRST pin to protect against spurious resets), and adding that capacitor breaks startup when the `HWBE` fuse is enabled (it effectively converts the power-on reset, which normally ignores the state of the HWB pin, into an external reset, which uses the HWB pin).

The `caterina` bootloader probably does not need the external PE2/HWB connection, because with that bootloader you need to disable the `HWBE` fuse and enable the `BOOTRST` fuse instead; however, that bootloader is very annoying for users, so you should not use it if you have any other options.

Maybe it could be possible to fix the `lufa-dfu`/`qmk-dfu` code to work correctly with the `BOOTRST` configuration, but that would require modifying the LUFA code (the bootloader sources actually live there).

MaxJ345, 2024-03-13:
I shorted pin **33** (`PE2`/`HWB`) to ground. I set the high fuse to `0x99` and extended fuse to `0xF3`. My results were:
* Physical reset (shorting the `RESET` pin to ground) --> bootloader mode
* `QK_BOOT` keycode --> bootloader mode

That solves my bootloader issue!

What is `BOOTRST`? What is `HWBE`? How are they related?

sigprof, 2024-03-13:
These are fuse bits, described in the [ATmega32U4 datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7766-8-bit-AVR-ATmega16U4-32U4_Datasheet.pdf):
[image]
[image]
[image]
(the documentation is not ideal — e.g., it mentions the ALE name for the HWB pin, but ATmega32U4 does not support the ALE function for that pin — that function is present in some other chips though).