qmk/qmk_firmware #23274: [Bug] error with command generate-compilation-database

ComteHerrapait, 2024-03-14:
### Describe the Bug

executing the command  `qmk generate-compilation-database -kb lily58 -km default` always result in the same error :
 ```console
Ψ Making clean with make clean
QMK Firmware 0.24.4
Deleting .build/ ... done.
Ψ Gathering build instructions from make -r -R -f builddefs/build_keyboard.mk -s -n KEYBOARD=lily58/rev1 KEYMAP=default KEYBOARD_FILESAFE=lily58_rev1 TARGET=lily58_rev1_default INTERMEDIATE_OUTPUT=.build/obj_lily58_rev1_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"
<class 'subprocess.CalledProcessError'>
☒ Command '['C:\\QMK_MSYS\\usr\\bin\\bash.exe', '-c', 'make -r -R -f builddefs/build_keyboard.mk -s -n KEYBOARD=lily58/rev1 KEYMAP=default KEYBOARD_FILESAFE=lily58_rev1 TARGET=lily58_rev1_default INTERMEDIATE_OUTPUT=.build/obj_lily58_rev1_default VERBOSE=false COLOR=true SILENT=false \'QMK_BIN="qmk"\'']' returned non-zero exit status 2.
Traceback (most recent call last):
  File "C:/QMK_MSYS/mingw64/lib/python3.11/site-packages/milc/milc.py", line 539, in __call__
    return self.__call__()
           ^^^^^^^^^^^^^^^
  File "C:/QMK_MSYS/mingw64/lib/python3.11/site-packages/milc/milc.py", line 544, in __call__
    return self._subcommand(self)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:/Projects/qmk_firmware/lib/python/qmk/decorators.py", line 27, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:/Projects/qmk_firmware/lib/python/qmk/decorators.py", line 47, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:/Projects/qmk_firmware/lib/python/qmk/cli/generate/compilation_database.py", line 143, in generate_compilation_database
    return target.generate_compilation_database()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/Projects/qmk_firmware/lib/python/qmk/build_targets.py", line 122, in generate_compilation_database
    ret = write_compilation_database(command=command, output_path=output_path, skip_clean=skip_clean, **env_vars)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/Projects/qmk_firmware/lib/python/qmk/cli/generate/compilation_database.py", line 104, in write_compilation_database
    result = cli.run(command, capture_output=True, check=True, env=env)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/QMK_MSYS/mingw64/lib/python3.11/site-packages/milc/milc.py", line 157, in run
    return subprocess.run(command, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/QMK_MSYS/mingw64/lib/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\QMK_MSYS\\usr\\bin\\bash.exe', '-c', 'make -r -R -f builddefs/build_keyboard.mk -s -n KEYBOARD=lily58/rev1 KEYMAP=default KEYBOARD_FILESAFE=lily58_rev1 TARGET=lily58_rev1_default INTERMEDIATE_OUTPUT=.build/obj_lily58_rev1_default VERBOSE=false COLOR=true SILENT=false \'QMK_BIN="qmk"\'']' returned non-zero exit status 2.
```

I tried : 
 * executing from the root of project with kb and km parameters
 * executing from the keymap folder with and without kb and km parameters
 * debugging the python file, it seems to try to execute the command `make -r -R -s -n KEYBOARD=lily58/rev1 KEYMAP=default KEYBOARD_FILESAFE=lily58_rev1 TARGET=lily58_rev1_default INTERMEDIATE_OUTPUT=.build/obj_lily58_rev1_def
ault VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"` which results in this error :  `make: *** No targets specified and no makefile found.  Stop.`

### Keyboard Used

lily56/rev1

### Operating System

Windows 11 Pro (23H2)

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: C:/Projects/qmk_firmware
Ψ Detected Windows 10 (10.0.22631).
Ψ QMK MSYS version: 1.8.0
Ψ Userspace enabled: False
Ψ Git branch: lily58-comte
Ψ Repo version: 0.24.4
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest lily58-comte: 2024-03-14 00:21:33 +0100 (1670177b3b) -- added custom keys media/mouse
Ψ - Latest upstream/master: 2024-03-13 08:24:21 +0100 (aa73362674) -- [Keyboard] add glyphkbd_v2 (#23131)
Ψ - Latest upstream/develop: 2024-03-13 23:04:49 +0000 (af1ac6b1bd) -- Reject duplicate matrix locations in LAYOUT macros (#23273)
Ψ - Common ancestor with upstream/master: 2024-03-13 08:24:21 +0100 (aa73362674) -- [Keyboard] add glyphkbd_v2 (#23131)
Ψ - Common ancestor with upstream/develop: 2024-03-13 08:24:21 +0100 (aa73362674) -- [Keyboard] add glyphkbd_v2 (#23131)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f90)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I am using QMK_MSYS on windows.

tzarc, 2024-03-14:
I'd hazard a guess and say there's something environmental at play:

```
[nickb@nickb-desktop ~]$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: E:/qmk_firmware
Ψ Detected Windows 10 (10.0.22621).
Ψ QMK MSYS version: 1.8.0
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.24.4
Ψ - Latest master: 2024-03-14 04:28:54 +0100 (1080025973) -- Update feature_tri_layer.md - typo (#23275)
Ψ - Latest upstream/master: 2024-03-14 04:28:54 +0100 (1080025973) -- Update feature_tri_layer.md - typo (#23275)
Ψ - Latest upstream/develop: 2024-03-14 03:57:48 +0000 (f1a279810f) -- Migrate features from rules.mk to DD (#23202)
Ψ - Common ancestor with upstream/master: 2024-03-14 04:28:54 +0100 (1080025973) -- Update feature_tri_layer.md - typo (#23275)
Ψ - Common ancestor with upstream/develop: 2024-03-14 04:28:54 +0100 (1080025973) -- Update feature_tri_layer.md - typo (#23275)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb16109)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f90)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```
```
[nickb@nickb-desktop ~]$ qmk generate-compilation-database -kb lily58 -km default
Ψ Making clean with make clean
QMK Firmware 0.24.4
Deleting .build/ ... done.
Ψ Gathering build instructions from make -r -R -f builddefs/build_keyboard.mk -s -n KEYBOARD=lily58/rev1 KEYMAP=default KEYBOARD_FILESAFE=lily58_rev1 TARGET=lily58_rev1_default INTERMEDIATE_OUTPUT=.build/obj_lily58_rev1_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"
Ψ Found 66 compile commands
Ψ Writing build database to E:/qmk_firmware/compile_commands.json
```

hitsmaxft, 2024-03-25:
I meet this issue if placing qmk_userspace under qmk_firmware. 

make will run error by missing `QMK_USERSPACE` variable 
`QMK_USERSPACE=/{PATH_TO_PROJECT}/qmk_firmware/qmk_userspace` 

Manually append to make command will work as expect
```shell
make -r -R -f builddefs/build_keyboard.mk -s -n KEYBOARD=annepro2/c18 KEYMAP=macvim KEYBOARD_FILESAFE=annepro2_c18 TARGET=annepro2_c18_macvim INTERMEDIATE_OUTPUT=.build/obj_annepro2_c18_macvim VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk" QMK_USERSPACE=/{MY_PATH}/qmk_firmware/qmk_userspace
```