qmk/qmk_firmware #23308: [Bug] mod tap bind is chording the tap key

XavierChanth, 2024-03-19:
### Describe the Bug

For some context:

Problem reveals itself when I'm using neovim inside tmux, set the chording/escape timeouts to 0 ms inside both neovim (ttimeout off) and tmux (escape-time). I have a binding in neovim which is `Meta+j`, `Meta+k`. And my option key is bound to chord ESC+ when held (terminal settings, not keyboard defined).

I have my caps-lock bound to `MT(MOD_LCTRL,KC_ESC)` so that tap is Escape and hold is LCTRL.
If I am moving really fast, I am able to chord `esc(physical caps key) + j`. I expect to only be able to chord `esc+j` when I am holding the normal escape key or the option key because of my software definition, not the caps key.

With a MT bind, I expect to only ever chord `LCTRL+j` not `ESC+j` or I expect two individual key presses of `ESC` then `j`.

### Keyboard Used

nuphy_air75v2 (see https://github.com/qmk/qmk_firmware/pull/22751)

### Link to product page (if applicable)

https://nuphy.com/collections/keyboards/products/air75-v2

### Operating System

macos 14.4 arm64 (m1 chip)

### qmk doctor Output

```text
➜ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /Users/chant/qmk_firmware
Ψ Detected macOS 14.4 (Apple Silicon).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.24.4
Ψ - Latest master: 2024-03-15 16:10:17 +1100 (c5f544cd96) -- Added WT65-H3 (HIBI HIBIKI) (#22886)
Ψ - Latest upstream/master: 2024-03-15 16:10:17 +1100 (c5f544cd96) -- Added WT65-H3 (HIBI HIBIKI) (#22886)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-03-15 16:10:17 +1100 (c5f544cd96) -- Added WT65-H3 (HIBI HIBIKI) (#22886)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f90)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

sigprof, 2024-03-19:
So your problem is that your program gets the press events for Esc and `j` without a delay in between, even though these were separate presses?

Tap events for dual-role keys like `MT()` are generated when the action is determined to be a tap, which usually happens only when the key had been released, and at that time the events which happened after the dual-role key was pressed are sent without any delay in between; the only delay that would be there is the USB polling interval (`USB_POLLING_INTERVAL_MS`), which is 1 ms by default.  So if you don't have `HOLD_ON_OTHER_KEY_PRESS` enabled, and the actual event sequence is
- press `MT(MOD_LCTRL,KC_ESC)`
- some delay
- press `KC_J`
- some delay (but the sum of those delays is below the tapping term)
- release `MT(MOD_LCTRL,KC_ESC)`

the keyboard will actually send the following events to the host when `MT(MOD_LCTRL,KC_ESC)` is released:
- press `KC_ESC`
- press `KC_J`
- delay for `TAP_CODE_DELAY` ms
- release `KC_ESC`

Note that there is no explicit delay between `KC_ESC` and `KC_J`, so the only delay there would be the USB polling interval (`USB_POLLING_INTERVAL_MS`), which is 1 ms by default, and any extra delay during processing may make that 1 ms delay effectively disappear.

As a workaround, you can try adding an extra delay in `post_process_record_user()`:
```c
void post_process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case MT(MOD_LCTRL,KC_ESC):
            if (record->event.pressed && (record->tap.count == 1)) {
                wait_ms(TAP_CODE_DELAY);
            }
            break;
    }
}
```
Maybe changing the core tapping code to insert `TAP_CODE_DELAY` immediately after reporting the press of the tap keycode instead of before release (as it does now) would make sense.  Although there is also a special case for `KC_CAPS` there, which uses the larger `TAP_HOLD_CAPS_DELAY`, and that code also exists in 3 copies (and one more without the `KC_CAPS` handling)…

XavierChanth, 2024-03-19:
I figured I would be looking at a workaround... thanks for giving me a starting point! I'll see what I come up with, although I currently see divergent branches for my board on the manufacturer's repo... so I'm hesitant to do anything until I'm sure which branch is safe to use (I would assume the most recent one with a PR open to this repo).

XavierChanth, 2024-03-19:
Part of the issue is the latency of the keyboard, I'm able to reproduce more easily in wireless mode. However, I can reduce the chances of unintentional chording by setting chording timeouts / delays to 0 in my shell / software, not going to bother modifying my keyboard firmware unless I absolutely have to.

Here's the changes I made to my configs for anyone curious:
bash/zsh: `set keyseq-timeout 0`
tmux: `set -sg escape-time 0`
nvim: `vim.opt.ttimeout = false`

closed 2024-03-19 by XavierChanth

XavierChanth, 2024-03-23:
For anyone that comes across this issue in the future, I found what I was looking for here:
https://github.com/qmk/qmk_firmware/blob/master/docs/tap_hold.md#hold-on-other-key-press

I was looking for the "hold on other key press" behavior for my tap-hold bind, not the default behavior.