qmk/qmk_firmware #23314: Logitech MX Keys S support

korpe1, 2024-03-20:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Would it be possible to ever support Logitech MX Keys S? Have anyone investigated this? If not possible, why?

fauxpark, 2024-03-20:
It's a wireless keyboard, and is not reprogrammable.

closed 2024-03-20 by fauxpark

korpe1, 2024-03-20:
> It's a wireless keyboard, and is not reprogrammable. QMK has specific hardware requirements, it won't run on just anything. Most mass produced keyboards are not compatible.

There are several, separate questions: 

1. Has the Logitech MX Keys S a non-locked bootloader or not? Can we replace it firmware by our own?

2. You say "It's a wireless keyboard, and is not reprogrammable.", what do you mean?
Most of wireless gadgets, peripherals, etc. are programmable/updatable and nothing itself in wirelessly technology doesn't prevent it. Also, AFAIK, Logitech MX Keys S support firmware updates by Logitech.

> QMK has specific hardware requirements, it won't run on just anything.
3. Yup, but if the points above are solved, we maybe can change this? AFAIK QMK have already wided the supported tech stack of it.

tzarc, 2024-03-20:
Feel free to do the research on all of your above points and report back. Once you have answers to those questions, we can then assess the feasibility of someone porting QMK to it.

Fair warning; none of the QMK team will do so -- it'll need to be community-led by people such as yourself.

korpe1, 2024-03-20:
> Fair warning; none of the QMK team will do so -- it'll need to be community-led by people such as yourself.

I understand this, but I have heard very talented people in this project and I wonder if anyone have any information or thoughts about this? I'm fairly new in embedded development. Could we keep one ticket open for this (e.g. flashing Logitech HID devices in general)?

And for the point two I would like to hear what @fauxpark meant? I'm new in these, but nevertheless that sounds weird.

tzarc, 2024-03-20:
Regarding wireless, this is our usual response on Discord:

> Currently Bluetooth/Wireless support within QMK Firmware is very limited. Consider using ZMK Firmware if wireless is your priority.
> 
> There are a few factors as to why this is the case - the main one tends to be wireless radio firmwares are generally incompatible with QMK's license, the GPL. This is usually due to regulatory constraints - QMK's license requires entire source code to be provided by manufacturers, which would also apply to any radio firmware if embedded within QMK. Radio firmwares are provided by manufacturers in binary form, thus cannot be compiled into QMK Firmware.
> 
> If the radio firmware were instead provided as source code, people could rebuild it and increase its transmission power, which would cause regulatory breaches - this is why radio firmwares are only provided as binaries.
> 
> Because of this fact, virtually any QMK-compatible Bluetooth/Wireless keyboard will use a secondary module which runs the wireless radio code external to QMK.

I'd be very surprised if the Logitech board in question would be any different.

As for not being reprogrammable -- it may technically be possible, but it's unlikely to be done with standard tooling available with QMK.

tzarc, 2024-03-20:
And there's no real point keeping the ticket open -- stalebot will automatically close it due to inactivity. If it's an unknown or incompatible microcontroller, adding support to the underlying libraries (before you even _think_ about QMK) may well take 6 months or more, even for skilled embedded engineers.

HorrorTroll, 2024-03-20:
> And there's no real point keeping the ticket open -- stalebot will automatically close it due to inactivity. If it's an unknown or incompatible microcontroller, adding support to the underlying [...]

I'd agree with this for closed, those Logitech wireless peripheral devices has been always using NRF52840. I've been searching and found the older MX Keys teardown, it revealed to be using NRF52840 too. So I don't think MX Keys S is different either.

korpe1, 2024-03-20:
> Regarding wireless, this is our usual response on Discord:

Thanks for info, but I can't understand that reason. I understand that if manufacturers want to use QMK and add drivers for wireless cards, them need to the drivers open-source. Anyway, AFAIK we can develop support for any propietary wireless card if we want to, right? Also Linux kernel with GPL license run on devices with proprietary firmwares.

korpe1, 2024-03-20:
> As for not being reprogrammable -- it may technically be possible, but it's unlikely to be done with standard tooling available with QMK. Very few mass-produced pieces of hardware use standard off-the-shelf microcontrollers specifically set up for DIY. Things are usually locked down to their own ecosystem, and if prior attempts with other pieces of hardware are anything to go by, you'll spend a significant length of time trying to even get proper access to write to flash. If it's a bootloader that requires signed firmware updates... you'll be out of luck.

Agree, but since Logitech can update them with Bluetooth and/or USB, also we should have able to do the same after flashing own firmware. We have able to flash own firmware if and only if the bootloader isn't locked.

korpe1, 2024-03-20:
> And there's no real point keeping the ticket open -- stalebot will automatically close it due to inactivity. If it's an unknown or incompatible microcontroller, adding support to the underlying [...]

The first thing to investigating is, are the bootloaders of Logitech HID devices locked or not? I think there might have be some volunteers interesting make this first step to use open-source firmware on them.

tzarc, 2024-03-20:
We eagerly await your PR.