qmk/qmk_firmware #23316: Two questions

korpe1, 2024-03-20:
> Regarding wireless, this is our usual response on Discord:
> 
> > Currently Bluetooth/Wireless support within QMK Firmware is very limited. Consider using ZMK Firmware if wireless is your priority.
> > There are a few factors as to why this is the case - the main one tends to be wireless radio firmwares are generally incompatible with QMK's license, the GPL. This is usually due to regulatory constraints - QMK's license requires entire source code to be provided by manufacturers, which would also apply to any radio firmware if embedded within QMK. Radio firmwares are provided by manufacturers in binary form, thus cannot be compiled into QMK Firmware.
> > If the radio firmware were instead provided as source code, people could rebuild it and increase its transmission power, which would cause regulatory breaches - this is why radio firmwares are only provided as binaries.
> > Because of this fact, virtually any QMK-compatible Bluetooth/Wireless keyboard will use a secondary module which runs the wireless radio code external to QMK.

Thanks for info, but I can't understand that reason. I understand that if manufacturers want to use QMK and add drivers for wireless cards, them need to the drivers open-source. Anyway, AFAIK we can develop support for any propietary wireless card if we want to, right? Also Linux kernel with GPL license run on devices with proprietary firmwares.

_Originally posted by @korpe1 in https://github.com/qmk/qmk_firmware/issues/23314#issuecomment-2008974182_
            

              > As for not being reprogrammable -- it may technically be possible, but it's unlikely to be done with standard tooling available with QMK. Very few mass-produced pieces of hardware use standard off-the-shelf microcontrollers specifically set up for DIY. Things are usually locked down to their own ecosystem, and if prior attempts with other pieces of hardware are anything to go by, you'll spend a significant length of time trying to even get proper access to write to flash. If it's a bootloader that requires signed firmware updates... you'll be out of luck.

Agree, but since Logitech can update them with Bluetooth and/or USB, also we should have able to do the same after flashing own firmware. We have able to flash own firmware if and only if the bootloader isn't locked.

_Originally posted by @korpe1 in https://github.com/qmk/qmk_firmware/issues/23314#issuecomment-2008978221_

tzarc, 2024-03-20:
Already resolved. Opening new issues here with the same content will result in a ban.

closed 2024-03-20 by tzarc

korpe1, 2024-03-20:
Sorry, I think you closed the issue because of my wrong wording. I don't assume the maintainers of QMK make free work for my intensions... Just trying open discussion around trying support mainstream products and the challenges in it. Apologies, I was rude :(

Anyway, can you clarify these two comments by you?

keyboard-magpie, 2024-03-20:
All your questions are searchable outside of github issues.

Open source bluetooth stacks are not feasible. The FCC and other regulatory bodies would require extensive testing which is not reasonable for most hobbyists. This is why current bluetooth solutions are provided as closed source binary files only.

Your second point is not a question. The replies on the other issue clarified this enough. QMK requires a compatible microcontroller with a compatible bootloader. These are all identified in the [documentation](https://docs.qmk.fm/#/compatible_microcontrollers?id=compatible-microcontrollers).

Again as was stated in the other issue, if the microcontroller is not supported, but is compatible, this could be added, but can take 6 months or more and extensive advanced programming experience.

I am going to lock this issue. If you want to continue discourse, I would politely suggest [Discord](https://discord.gg/qmk) over Github.