qmk/qmk_firmware #23347: [Bug] Tofu60 unable to wake from sleep on Linux

Limero, 2024-03-25:
### Describe the Bug

I recently purchased a Tofu60 2.0 and with the stock QMK firmware I had trouble that the keyboard would wake the computer from sleep, but then stop functioning until it was unplugged and plugged in again.

In an attempt to resolve this I built QMK from scratch (the product id changed from OG60 to TOFU60, so stock firmware must've been built with an old version). The problem now is that the keyboard is unable to wake the computer from sleep, but once woken with the computer power button, the keyboard functions as intended.

This issue is possibly only affecting Linux, as I'm able to wake a Mac from sleep. I'm not able to test with Windows.

Same issue happens with the [pre-built via firmware](https://www.caniusevia.com/docs/download_firmware)

### Keyboard Used

dztech/tofu60

### Operating System

Fedora 39. Linux 6.7.9

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

KarlK90, 2024-03-26:
Hello @Limero PR #21656 landed in the `develop` branch some weeks ago that addresses wake-up issues on ChibiOS based boards (tofu60 is based on the rp2040 it seems, so it uses ChibiOS under the hood). Please compile a firmware from the `develop` branch and see if this solves the issues for you.

KarlK90, 2024-04-01:
> @KarlK90 Thanks! Using `develop` did fix this particular issue 

@Limero Great that it solved your issue. 

> but caused another. The key used to wake the computer goes directly to my lock screen ([swaylock](https://github.com/swaywm/swaylock)), meaning that if I unlock the computer with a letter key, it has to be the first letter of my password or I will need to backspace. This issue doesn't seem to happen on macOS nor on any of my other keyboards.

This is actually a bug in QMK as the key presses used to wake up the host are not suppressed. 

Before the new async usb endpoints it was a bit of a gamble depending on keyboard MCU and OS if the key press used to wake the host actually reached the host. For the new async USB handling, which should guarantee the delivery, I just implemented #23389 which suppresses all wakeup key presses. Please try if this fixes the issue.