qmk/qmk_firmware #23349: [Bug] Modifier should be sent before regular key

ibash, 2024-03-25:
### Describe the Bug

Steps:
1. Turn on console + matrix debugging
2. Press "cmd" + "a" simultaneously

Expected:
1. "cmd" gets reported first and then "a" gets reported (so that the OS recognizes "cmd+a" as select all)

Actual:
1. "cmd" and "a" gets reported at the same time, so macOS doesn't recognize it

### Keyboard Used

keychron/q3/ansi

### Operating System

14.4 (23E214)

### qmk doctor Output

```text
❯  qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /Users/islam/Projects/qmk_firmware
Ψ Detected macOS 14.4 (Apple Silicon).
Ψ Git branch: keychron-hc595
Ψ Repo version: 0.19.4
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest keychron-hc595: 2024-03-18 13:17:41 -0700 (6bed9f600c0) -- Per key debouncing at 5ms
Ψ - Latest upstream/master: 2023-06-24 11:54:23 -0700 (3a42fbb0c9f) -- Mechlovin Zed65 rev1 Touch-Up (#21373)
Ψ - Latest upstream/develop: 2023-06-26 11:36:32 +0300 (3ebdb1258bc) -- Chibios USB: Take into account if host wants remote wakeup or not (#21287)
Ψ - Common ancestor with upstream/master: 2022-12-15 05:19:39 -0300 (33cc81f36ef) -- [Keymap] Simplifying ortho-4x12 jotix's layout (#19336)
Ψ - Common ancestor with upstream/develop: 2022-12-15 05:19:39 -0300 (33cc81f36ef) -- [Keymap] Simplifying ortho-4x12 jotix's layout (#19336)
Ψ CLI installed in virtualenv.
☒ Can't find avr-gcc in your path.
Would you like to install dependencies? [Y/n] y
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I am using keychron's branch, this exact commit: https://github.com/ibash/qmk_firmware/tree/keychron-hc595

ibash, 2024-03-25:
Here's a screenshot of the keymap log, cmd + a is sent in the same report but not detected as a "select-all" from the macOS side

[image]

sigprof, 2025-07-06:
QMK does not explicitly apply any priority to the key state changes detected in the same matrix scan — it just processes matrix rows in the increasing order, and processes columns within each row also in the increasing order:
https://github.com/qmk/qmk_firmware/blob/9c965bb62ec9ea84e68e0a1559dfbc35429df090/quantum/keyboard.c#L614-L636
Usually this is not a problem, because timing the physical press actions for multiple keys so that they fall into the same matrix scan is expected to be rare.  However, in the @ibash case the debugging code may slow down things enough to make that possible, and in the @Atemu case the Ergodox EZ board is known to have a low scan rate (the I2C expander is slow).  Another thing which could contribute to the problem is the default `sym_defer_g` debouncing algorithm, which would combine any key state changes which happened within the debounce interval into a single matrix state change; however, apparently in the @ibash case `sym_defer_pk` is used, which should not have that problem.

Not sure what could be done to solve this problem in a generic way (the matrix may be wired in an arbitrary way, and the modifier or layer switch keys could be placed anywhere).