qmk/qmk_firmware #23440: [Bug] warning:The official repository does not seem to be configured as git remote "upstream".

uhkkhilhih, 2024-04-08:
### Describe the Bug

[1***@LAPTOP-**** ~]$ qmk setup                                              
Ψ Found qmk_firmware at C:/Users/18***/qmk_firmware.                              
Ψ QMK Doctor is checking your environment.                                        
Ψ CLI version: 1.1.5                                                              
Ψ QMK home: C:/Users/18***/qmk_firmware                                           
Ψ Detected Windows 10 (10.0.22631).                                               
Ψ QMK MSYS version: 1.9.0                                                         
Ψ Userspace enabled: False                                                        
Ψ Git branch: master                                                              
Ψ Repo version: 0.24.7                                                            
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.                                                 
Ψ Found arm-none-eabi-gcc version 12.2.0                                          
Ψ Found avr-gcc version 12.2.0                                                    
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.    
Ψ Found avrdude version 7.0                                                       
Ψ Found dfu-programmer version 1.1.0                                              
Ψ Found dfu-util version 0.11                                                     
Ψ Submodules are up to date.                                                      
Ψ Submodule status:                                                               
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)                        
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f90)                 
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)                      
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)                           
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)                             
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)                           
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)                         
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)                            
Ψ QMK is ready to go, but minor problems were found                               

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

how to solve this problem?

fauxpark, 2024-04-08:
This is not a bug. You are using some other fork of QMK, not `qmk/qmk_firmware`. `git remote -v` will show you.

closed 2024-04-08 by fauxpark

uhkkhilhih, 2024-04-08:
> This is not a bug. You are using some other fork of QMK, not `qmk/qmk_firmware`. `git remote -v` will show you.

I have asked the GPT and tried "git remote -v",but it didn't work.like this:
git remote -v
origin  https://github.com/qmk/qmk_firmware.git (fetch)
origin  https://github.com/qmk/qmk_firmware.git (push)
upstream        https://github.com/qmk/qmk_firmware.git (fetch)
upstream        https://github.com/qmk/qmk_firmware.git (push)

uhkkhilhih, 2024-04-08:
GPT told me to input：
git remote rename upstream origin
git remote add upstream https://github.com/qmk/qmk_firmware.git
but the warning didn't disappear

fauxpark, 2024-04-08:
Stop using ChatGPT. It will not help you.

Delete your repo and run `qmk setup` again.

uhkkhilhih, 2024-04-08:
> Stop using ChatGPT. It will not help you.
> 
> Delete your repo and run `qmk setup` again.
sorry,but how to delete my repo?

fauxpark, 2024-04-08:
Do you not know how to delete a folder?

uhkkhilhih, 2024-04-08:
> Do you not know how to delete a folder?
maybe just press the delete key to delete the folder?but I dont know where the repo is

fauxpark, 2024-04-08:
`qmk setup` already told you where it is.

uhkkhilhih, 2024-04-08:
> `qmk setup` already told you where it is. You already know this because you have censored your username.

does the repo in the qmk firmware folder?

fauxpark, 2024-04-08:
Yes.

uhkkhilhih, 2024-04-08:
> Yes.
does the "repo" means repository?
I searched "repo" and only find two documents named "user_repo_v1.jsonschema" and "user_repo_v0.jsonschema",should I delete them?
(there are some other documents named "report" but I dont think they are the right documents I need.)

keyboard-magpie, 2024-04-08:
Please seek further help on discord if absolutely required https://discord.gg/qmk 

Locking this