qmk/qmk_firmware #23523: Keys are held down and keep repeating after waking from sleep.

crimsonyote, 2024-04-14:
### Describe the Bug

Like I said in the title. I have an AKKO 5075S VIA board. This issue appears whenever my laptop goes into sleep mode with the board connected. I will go to press a key to wake it again, to find that the same key is held down and spamming itself into the password box. The only way to fix this is to press another key and delete what it's typed.

### Keyboard Used

AKKO 5075S VIA

### Operating System

Windows 11

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

griffingarland, 2024-07-13:
I believe I am having a similar issue. I am using a [Keeb.io sinc split keyboard](https://keeb.io/products/sinc-keyboard-pre-built). It uses RP2040s for the microcontroller.

I was having trouble waking the computer from sleep by typing on the keyboard, which I was able to do with my previous keyboard I used on this computer. I found [this github issue, was fixed last year](https://github.com/qmk/qmk_firmware/issues/19008) So I figured I needed to flash the keyboard firmware. 

Today I downloaded QMK MSys (currently using qmk-cli version 1.1.5 which appears to the be latest released version) and flashed the keyboard. After this, after hitting a key on the keyboard will repeatedly type that key (I disabled the password requirement when entering Windows 11 to confirm this). So, for example, if I wake my computer up by hitting "j" it will repeatedly type "j" once entering Windows 11.

I can't seem to get it to stop unless I unplug and replug in the keyboard.
I tried compiling the firmware with various [config options](https://github.com/qmk/qmk_firmware/blob/master/docs/config_options.md) such as NO_USB_STARTUP_CHECK, USB_WAIT_FOR_ENUMERATION, and setting USB_SUSPEND_WAKEUP_DELAY to numbers ranging from 200 to 5000. None of it worked.

I should mention that the keyboard is plugged into a [USB switch](https://www.amazon.com/gp/product/B08MBXMZLV) (basically just a hub with a button that allows me to switch between different computers). I plug all my peripherals into it and use it to switch my peripherals between my work and personal computers. When I plug directly into my desktop via USB, it doesn't seem to happen. It seems to be exclusive to using the USB switch. However, it only happens when waking up Windows by typing on the keyboard, if I use the mouse or power button to wake up the computer it doesn't happen.

KarlK90, 2024-07-14:
@griffingarland could you post the output of the `qmk doctor` command?

griffingarland, 2024-07-15:
```
[Griffin@Griffins-Desktop ~]$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: C:/Users/Griffin/qmk_firmware
Ψ Detected Windows 10 (10.0.22631).
Ψ QMK MSYS version: 1.9.0
Ψ Userspace enabled: False
Ψ Git branch: develop
Ψ Repo version: 0.25.17
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest develop: 2024-07-13 14:06:19 -0400 (484edf4449) -- Add my own configuration
Ψ - Latest upstream/master: 2024-07-13 23:40:36 +1000 (875eb43a28) -- `dp60`: move RGB Matrix LED config to data driven (#24022)
Ψ - Latest upstream/develop: 2024-07-15 08:42:14 +0000 (fef8e7195b) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2024-07-12 22:11:00 -0400 (7b5d7907df) -- adds veilid SAO macropad (#23868)
Ψ - Common ancestor with upstream/develop: 2024-07-13 18:41:05 +1000 (1552cf2ddc) -- Update atomic GPIO macros in keyboard custom matrix (#23796)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

Nuc1eoN, 2024-10-09:
I have the same issue as @crimsonyote on a xiudi xd75.

Jinra, 2025-04-08:
My Canoe gen2 does this, specifically for my left control (no matter what is mapped to it), but is quickly resolved if i just press ESC

perrin4869, 2025-04-16:
I updated my dactyl_manuform after a few years, and I ran into this issue

perrin4869, 2025-10-20:
I could fix this by using `#define USB_SUSPEND_WAKEUP_DELAY 500`, I'm on `qmk 0.30.5`