qmk/qmk_firmware #23531: [Bug] If 4 rotary encoders are specified, all rotary encoders will stop responding.

Taro-Hayashi, 2024-04-15:
### Describe the Bug

If 4 rotary encoders are specified, all rotary encoders will stop responding. If 3 or 5 rotary encoders are specified, they work as usual. I chose this board as an easy example to try, but confirmed that it occurs with ATmega32u4 and RP2040.

It's not work
~~~
    "encoder": {
        "rotary": [
            {"pin_a": "B6", "pin_b": "B2"},
            {"pin_a": "B3", "pin_b": "B1"},
            {"pin_a": "B3", "pin_b": "B1"},
            {"pin_a": "B3", "pin_b": "B1"}
        ]
    },
~~~

It's work
~~~
    "encoder": {
        "rotary": [
            {"pin_a": "B6", "pin_b": "B2"},
            {"pin_a": "B3", "pin_b": "B1"},
            {"pin_a": "B3", "pin_b": "B1"}
        ]
    },
~~~

~~~
    "encoder": {
        "rotary": [
            {"pin_a": "B6", "pin_b": "B2"},
            {"pin_a": "B3", "pin_b": "B1"},
            {"pin_a": "B3", "pin_b": "B1"},
            {"pin_a": "B3", "pin_b": "B1"},
            {"pin_a": "B3", "pin_b": "B1"}
        ]
    },
~~~
Thank you.

### Keyboard Used

25keys/cassette42

### Operating System

macOS Sonoma 14.4.1

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /Users/hayashi/qmk_firmware_raw
Ψ Detected macOS 14.4.1 (Apple Silicon).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.24.7
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f90)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

fauxpark, 2024-04-16:
I'm not really sure how you expect this to work with multiple encoders on the same pins...

Taro-Hayashi, 2024-04-16:
Sorry about the example. It does not matter which pins are specified, just that I wanted to avoid using a pin that is already used. 
If any other pins are specified, it will not work if there are 4 rotary encoders.

tokolist, 2024-04-23:
I faced identical issue. If there is more than three encoders configured, all encoders stop working, It doesn't matter which pins are specified.

jaygreco, 2024-04-23:
We ran into this on the TIDBIT (it’s still reproducible). I bisected down to https://github.com/qmk/qmk_firmware/commit/3e1ac7a38fa4e6885053a762bc75f7c4e068eccb 

before this commit, all is well. Anything after this and the encoders break. I wasn’t able to debug further unfortunately.

sigprof, 2024-04-23:
Please try #23595 — apparently `encoder_queue_full_advanced()` was broken when `MAX_QUEUED_ENCODER_EVENTS` was not a power of 2, which happened when `NUM_ENCODERS_MAX_PER_SIDE` increased above 3.

Taro-Hayashi, 2024-04-23:
Hi, I tried it and my problem was solved. I hope the problems of others have also been resolved.
i appreciate your help. I will close this one.

closed 2024-04-23 by Taro-Hayashi